coolsnowwolf/lede #3: 大神，編譯X86，保存掛載點出錯

ctw8989, 2017-09-06:
/usr/lib/lua/luci/util.lua:623: Unable to establish ubus connection stack traceback: [C]: in function 'assert' /usr/lib/lua/luci/util.lua:623: in function 'ubus' /usr/lib/lua/luci/dispatcher.lua:172: in function 'session_retrieve' /usr/lib/lua/luci/dispatcher.lua:382: in function 'dispatch' /usr/lib/lua/luci/dispatcher.lua:121: in function

官方的也一樣

coolsnowwolf, 2017-09-07:
切换回到 官方 17.01.2 的 feeds 了，你可以再次 feed update 编译一次看看

ctw8989, 2017-09-07:
大哥，刚试了下，还是不行，保存后还会出错
编译3800，4300这些都没问题
不会是和我用Ubuntu  16 编译有关吧

coolsnowwolf, 2017-09-07:
亲测没有问题，14 LTS 编译.

LUA 依赖你的 local，请使用合适的版本编译

ctw8989, 2017-09-08:
那就奇怪了，我又换了虚拟机，也按照你的方法建编译环境啦，死活都不行，出错后连WEB都进不了
提示如下：
/usr/lib/lua/luci/dispatcher.lua:478: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/util.lua:623: Unable to establish ubus connection
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:478: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

而且在adbyby页面点  重启adbyby   时，也提示：
Bad Gateway
The process did not produce any response

ctw8989, 2017-09-09:
@coolsnowwolf  L大，我真是服了，无论如何试，也试过恩山上的所谓菜鸟包，还是一样。你看能不能把你的编译x86_64的 .config文件发我一份，估计是有什么包我没选上！拜托了。
ctw8989@gmail.com   谢谢！！

ctw8989, 2017-09-09:
XX了，原来是我刷错固件了

coolsnowwolf, 2017-09-10:
好吧~

closed 2017-09-10 by coolsnowwolf