coolsnowwolf/lede #7: 借用您的软件源码用在lede17.01.2，打了高通fast path补丁,编译卡壳

tomerilery, 2017-09-07:
用了您的软件源码,package/lean, 自己克隆了lede17.01.2，也打了高通fast path补丁，下面是打补丁过程：
wget https://github.com/lede-project/source/commit/b87b4734c6e56fa45ec612350e2aa480ed2d8dd6.patch
patch -p1 < b87b4734c6e56fa45ec612350e2aa480ed2d8dd6.patch
（略过了4.9内核）
mv target/linux/generic/hack-4.4 patches-4.4
Kernel modules > Network Support >kmod-fast-classifier，kmod-shortcut-fe，kmod-shortcut-fe-cm选中
然后编译的过程报错如下：
make[3]: Entering directory `/home/mltao/lede17.01.2/source/package/kernel/shortcut-fe'
make -C "/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/linux-4.4.86" ARCH="arm" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" SUBDIRS="/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe" EXTRA_CFLAGS="" modules
make[4]: Entering directory `/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/linux-4.4.86'
  CC [M]  /home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe/sfe_ipv4.o
/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_udp':
/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe/sfe_ipv4.c:1366:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^
/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_tcp':
/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe/sfe_ipv4.c:1909:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^
make[5]: *** [/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe/sfe_ipv4.o] Error 1
make[4]: *** [_module_/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe] Error 2
make[4]: Leaving directory `/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/linux-4.4.86'
make[3]: *** [/home/mltao/lede17.01.2/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/shortcut-fe/.built] Error 2
make[3]: Leaving directory `/home/mltao/lede17.01.2/source/package/kernel/shortcut-fe'
make[2]: *** [package/kernel/shortcut-fe/compile] Error 2
make[2]: Leaving directory `/home/mltao/lede17.01.2/source'
make[1]: *** [/home/mltao/lede17.01.2/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/mltao/lede17.01.2/source'
make: *** [world] 错误 2
请问怎么解决？谢谢！
来自石像鬼群：排排

coolsnowwolf, 2017-09-07:
目录名字不同，17.1.02 在 4.4 目录中

closed 2017-09-07 by coolsnowwolf

yingriyanlong, 2017-10-18:
我也遇到了同样的问题，大佬可以说详细一些吗

vincentchen, 2018-03-12:
大雕，用你源码内核改4.14也报这个错

xiazeyu, 2021-01-02:
> 目录名字不同，17.1.02 在 4.4 目录中

是手动改 https://github.com/lede-project/source/commit/b87b4734c6e56fa45ec612350e2aa480ed2d8dd6.patch 这个patch文件里的路径吗

xiazeyu, 2021-01-02:
贴一个支持5.4内核的sfe https://github.com/MeIsReallyBa/Openwrt-sfe-flowoffload-linux-5.4