coolsnowwolf/lede #11: 编译错误，貌似pdnsd-1.2.9a-par，但没找到dl文件夹......

ziqqqq, 2017-09-08:
pdnsd-1.2.9a-par;     '
Checking out files from the git repository...
Cloning into 'pdnsd-1.2.9a-par'...
fatal: unable to connect to gitorious.org:
gitorious.org[0: 64.13.172.37]: errno=Connection refused

Makefile:55: recipe for target '/home/zh/lede/dl/pdnsd-1.2.9a-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz' failed
make[3]: *** [/home/zh/lede/dl/pdnsd-1.2.9a-par-a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d.tar.gz] Error 128
make[3]: Leaving directory '/home/zh/lede/package/lean/pdnsd-alt'
package/Makefile:109: recipe for target 'package/lean/pdnsd-alt/compile' failed
make[2]: *** [package/lean/pdnsd-alt/compile] Error 2
make[2]: Leaving directory '/home/zh/lede'
package/Makefile:105: recipe for target '/home/zh/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/zh/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zh/lede'
/home/zh/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2

daofeng2015, 2017-09-08:
更新源码就有了

ziqqqq, 2017-09-08:
更新过了........

coolsnowwolf, 2017-09-08:
gitorious.org 这个网址可能过期了，我 fork 了一份在本地，你 git pull 一下吧

ziqqqq, 2017-09-08:
谢谢，已找到......

closed 2017-09-08 by coolsnowwolf