coolsnowwolf/lede #16: mvebu平台编译问题

MeISbug, 2017-09-08:
你好，

到了最后阶段，出现以下错误:

Image Name:
Created:      Thu Sep  7 15:02:20 2017
Image Type:   ARM Linux Script (uncompressed)
Data Size:    848 Bytes = 0.83 kB = 0.00 MB
Load Address: 00000000
Entry Point:  00000000
Contents:
   Image 0: 840 Bytes = 0.82 kB = 0.00 MB
rm -f /home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/tmp/lede-mvebu-armada-388-clearfog-pro-sdcard.img.gz.boot
mkfs.fat -C /home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/tmp/lede-mvebu-armada-388-clearfog-pro-sdcard.img.gz.boot 16384
mkfs.fat 4.1 (2017-01-24)
mcopy -i /home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/tmp/lede-mvebu-armada-388-clearfog-pro-sdcard.img.gz.boot /home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/linux-4.4.85/arch/arm/boot/dts/armada-388-clearfog-pro.dtb ::armada-388-clearfog-pro.dtb;  mcopy -i /home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/tmp/lede-mvebu-armada-388-clearfog-pro-sdcard.img.gz.boot /home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/linux-4.4.85/arch/arm/boot/dts/armada-388-clearfog-base.dtb ::armada-388-clearfog-base.dtb;
/home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/linux-4.4.85/arch/arm/boot/dts/armada-388-clearfog-pro.dtb: No such file or directory
/home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/linux-4.4.85/arch/arm/boot/dts/armada-388-clearfog-base.dtb: No such file or directory
make[5]: *** [/home/test/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu/tmp/lede-mvebu-armada-388-clearfog-pro-sdcard.img.gz] Error 1
make[5]: Leaving directory `/home/test/lede/target/linux/mvebu/image'
make[4]: *** [install] Error 2
make[4]: Leaving directory `/home/test/lede/target/linux/mvebu'
make[3]: *** [install] Error 2
make[3]: Leaving directory `/home/test/lede/target/linux'
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory `/home/test/lede'
make[1]: *** [/home/test/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory `/home/test/lede'
make: *** [world] Error 2

请问应该怎么解决？

另外发现到了这步，/lede/bin/targets/mvebu/generic/下已经有固件生成，但是刷机后路由器无法启动，网卡获取不了IP地址，是因为前面的错误造成的吗？

谢谢！

coolsnowwolf, 2017-09-08:
看到了，缺少 dtd，我来修复下

coolsnowwolf, 2017-09-08:
确实能启动

你那边有 ttl 启动日志么？我看看原因

MeISbug, 2017-09-08:
这个要拆机才行，而且我这边目前没有TTL线。

我试过X86的版本是可以启动的，刚刚看到有人用EA8500编译也能启动，看来问题是mvebu的。

之前是用本地电脑的虚拟机编译，路由器SSR开了白名单模式，明天试试用全局，我想会不会是uboot之类下载不成功的原因？

MeISbug, 2017-09-09:
已经拆机，并问朋友借TTL线，过两天提供启动日志

coolsnowwolf, 2017-09-10:
目前还是有问题，在哎mvebu 上面，应该是官方某个 trunk 的时候引入的错误

我会另外起个 lede-17.01 正式版的分支，那个没问题

coolsnowwolf, 2017-09-10:
等你的 TTL 信息再看看

MeISbug, 2017-09-10:
大神，我终于拿到TTL信息了, 这个是不能启动的:

BootROM - 1.73
Booting from NAND flash

General initialization - Version: 1.0.0
AVS selection from EFUSE disabled (Skip reading EFUSE values)
mvSysEnvIsFlavourReduced: TWSI Read of 'flavor' failed
Detected Device ID 6820
High speed PHY - Version: 2.0

Init RD NAS topology Serdes Lane 3 is USB3
Serdes Lane 4 is SGMII
board SerDes lanes topology details:
 | Lane # | Speed|    Type     |
 ------------------------------|
 |   0    |  6   |  SATA0      |
 |   1    |  5   |  PCIe0      |
 |   2    |  6   |  SATA1      |
 |   3    |  5   |  USB3 HOST1 |
 |   4    |  5   |  PCIe1      |
 |   5    |  0   |  SGMII2     |
 -------------------------------
:** Link is Gen1, check the EP capability
PCIe, Idx 0: Link upgraded to Gen2 based on client cpabilities
:** Link is Gen1, check the EP capability
PCIe, Idx 1: remains Gen1
High speed PHY - Ended Successfully
DDR3 Training Sequence - Ver TIP-1.46.0
mvSysEnvGetTopologyUpdateInfo: TWSI Read failed
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
DDR3 Training Sequence - Ended Successfully
Not detected suspend to RAM indication
BootROM: Image checksum verification PASSED

 __   __                      _ _
|  \/  | __ _ _ ____   _____| | |
| |\/| |/ _` | '__\ \ / / _ \ | |
| |  | | (_| | |   \ V /  __/ | |
|_|  |_|\__,_|_|    \_/ \___|_|_|
         _   _     ____              _
        | | | |   | __ )  ___   ___ | |_
        | | | |___|  _ \ / _ \ / _ \| __|
        | |_| |___| |_) | (_) | (_) | |_
         \___/    |____/ \___/ \___/ \__|
 ** LOADER **

U-Boot 2013.01 (Apr 08 2016 - 15:47:50) Marvell version: 2015_T1.QA.0p16

Boot version : v0.0.4

Board: RD-NAS-88F6820-DDR3
SoC:   MV88F6820 Rev A0
       running 2 CPUs
CPU:   ARM Cortex A9 MPCore (Rev 1) LE
       CPU 0
       CPU    @ 1866 [MHz]
       L2     @ 933 [MHz]
       TClock @ 200 [MHz]
       DDR3    @ 933 [MHz]
       DDR3 32 Bit Width,FastPath Memory Access, DLB Enabled, ECC Disabled
DRAM:  512 MiB
NAND:  256 MiB
MMC:   mv_sdh: 0
DEVINFO offset == 0x7e0000
U-ENV offset == 0x200000
U-ENV offset == 0x200000
S-ENV offset == 0x220000

#### auto_recovery ####
[u_env] get auto_recovery == yes
[u_env] get auto_recovery == yes
[u_env] get boot_part == 1
[u_env] get boot_part_ready == 3
auto_recovery enabled:1, boot_part:1, boot_part_ready:3

S-ENV offset == 0x220000
[boot_count_read] block:0x220000, size:128KB, records:64
[boot_count_read_record] boot_count:1, next_record:50

[boot_count_write] erase:0, auto_recovery->block_offset:0x220000 offset=0x239000

Updating boot_count ...
[boot_count_write] offset:0x239000 , length:2048
done

PCI-e 0 (IF 0 - bus 0) Root Complex Interface, Detected Link X1, GEN 2.0
PCI-e 1 (IF 1 - bus 1) Root Complex Interface, Detected Link X1, GEN 1.1
USB2.0 0: Host Mode
USB3.0 1: Host Mode
USB3.0 0: Host Mode

Map:   Code:                    0x1feab000:0x1ff75a20
       BSS:                     0x1ffef164
       Stack:                   0x1f9aaf20
       Heap:                    0x1f9ab000:0x1feab000
U-ENV offset == 0x200000
U-ENV offset == 0x200000
       U-Boot Environment:      0x00200000:0x00220000 (NAND)

Board configuration detected:
mvEthE6171SwitchBasicInit init
Net:
|  port  | Interface | PHY address  |
|--------|-----------|--------------|
| egiga0 |   RGMII   |     0x01     |
| egiga1 |   SGMII   |     0x00     |
| egiga2 |   SGMII   |   In-Band    |
egiga0 [PRIME], egiga1, egiga2
auto_recovery_check changes bootcmd: run nandboot
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0xa00000, size 0x600000
 6291456 bytes read: OK
## Booting kernel from Legacy Image at 02000000 ...
   Image Name:   ARM LEDE Linux-4.4.85
   Created:      2017-09-07  15:02:20 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2333374 Bytes = 2.2 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
OK

Starting kernel ...

卡在kenel这就不动了，等了两三分钟，没有后续的信息出来

MeISbug, 2017-09-10:
这个是用LEDE官方源码编译的，启动正常:

▒▒EFUSE disabled (Skip reading EFUSE values)
mvSysEnvIsFlavourReduced: TWSI Read of 'flavor' failed
Detected Device ID 6820
High speed PHY - Version: 2.0

Init RD NAS topology Serdes Lane 3 is USB3
Serdes Lane 4 is SGMII
board SerDes lanes topology details:
 | Lane # | Speed|    Type     |
 ------------------------------|
 |   0    |  6   |  SATA0      |
 |   1    |  5   |  PCIe0      |
 |   2    |  6   |  SATA1      |
 |   3    |  5   |  USB3 HOST1 |
 |   4    |  5   |  PCIe1      |
 |   5    |  0   |  SGMII2     |
 -------------------------------
:** Link is Gen1, check the EP capability
PCIe, Idx 0: Link upgraded to Gen2 based on client cpabilities
:** Link is Gen1, check the EP capability
PCIe, Idx 1: remains Gen1
High speed PHY - Ended Successfully
DDR3 Training Sequence - Ver TIP-1.46.0
mvSysEnvGetTopologyUpdateInfo: TWSI Read failed
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
DDR3 Training Sequence - Ended Successfully
Not detected suspend to RAM indication
BootROM: Image checksum verification PASSED

 __   __                      _ _
|  \/  | __ _ _ ____   _____| | |
| |\/| |/ _` | '__\ \ / / _ \ | |
| |  | | (_| | |   \ V /  __/ | |
|_|  |_|\__,_|_|    \_/ \___|_|_|
         _   _     ____              _
        | | | |   | __ )  ___   ___ | |_
        | | | |___|  _ \ / _ \ / _ \| __|
        | |_| |___| |_) | (_) | (_) | |_
         \___/    |____/ \___/ \___/ \__|
 ** LOADER **

U-Boot 2013.01 (Apr 08 2016 - 15:47:50) Marvell version: 2015_T1.QA.0p16

Boot version : v0.0.4

Board: RD-NAS-88F6820-DDR3
SoC:   MV88F6820 Rev A0
       running 2 CPUs
CPU:   ARM Cortex A9 MPCore (Rev 1) LE
       CPU 0
       CPU    @ 1866 [MHz]
       L2     @ 933 [MHz]
       TClock @ 200 [MHz]
       DDR3    @ 933 [MHz]
       DDR3 32 Bit Width,FastPath Memory Access, DLB Enabled, ECC Disabled
DRAM:  512 MiB
NAND:  256 MiB
MMC:   mv_sdh: 0
DEVINFO offset == 0x7e0000
U-ENV offset == 0x200000
U-ENV offset == 0x200000
S-ENV offset == 0x220000

#### auto_recovery ####
[u_env] get auto_recovery == yes
[u_env] get auto_recovery == yes
[u_env] get boot_part == 2
[u_env] get boot_part_ready == 3
[... 333 lines cut ...]
[   14.403581] mwifiex_sdio mmc0:0001:1: info: MWIFIEX VERSION: mwifiex 1.0 (15.68.7.p5)
[   14.403584] mwifiex_sdio mmc0:0001:1: driver_version = mwifiex 1.0 (15.68.7.p5)
[   14.421112] device eth0 entered promiscuous mode
[   14.429727] br-lan: port 1(eth0) entered forwarding state
[   14.435191] br-lan: port 1(eth0) entered forwarding state
[   14.448042] mvneta f1070000.ethernet eth1: configuring for fixed link mode
[   14.455486] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   14.461616] mvneta f1070000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[   14.469520] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   15.256504] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   15.545505] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   15.551596] device wlan1 entered promiscuous mode
[   15.556355] br-lan: port 2(wlan1) entered forwarding state
[   15.561886] br-lan: port 2(wlan1) entered forwarding state
[   15.567510] br-lan: port 2(wlan1) entered disabled state
[   15.573191] device wlan0 entered promiscuous mode
[   15.788325] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   15.794755] br-lan: port 2(wlan1) entered forwarding state
[   15.800279] br-lan: port 2(wlan1) entered forwarding state
[   16.337393] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   16.343823] br-lan: port 3(wlan0) entered forwarding state
[   16.349344] br-lan: port 3(wlan0) entered forwarding state
[   16.431323] br-lan: port 1(eth0) entered forwarding state
[   17.791307] br-lan: port 2(wlan1) entered forwarding state
[   18.341309] br-lan: port 3(wlan0) entered forwarding state

MeISbug, 2017-09-12:
刚刚编译了7.4的版本，可以正常启动了，准备愉快的玩耍～～  感谢！

coolsnowwolf, 2017-09-12:
已修复

closed 2017-09-12 by coolsnowwolf