coolsnowwolf/lede #21: 一直编译出，小白一个也不知道是哪出了问题

jackyan999, 2017-09-08:
一直严格按教程，二天都没编译成一个，换了好几个硬件环境也都同样，提示错误1，也在谷歌上找了资料，说是网络的问题，但是网格一直都是好的，也启动了SS服务，哪位老司机知道，麻烦引导。谢谢（用make V=99 2>&1 | tee build.log | grep -i error命令查出的红色字体不知道是什么提示）
[image]

jackyan999, 2017-09-08:
[image]

jackyan999, 2017-09-08:
[image]

MeISbug, 2017-09-08:
你先加上-j1参数吧，会有详细的出错信息。

截图也要截完整啊，图1，2都不全（虽然1，2没有什么有用信息）

jackyan999, 2017-09-08:
[image]
问题终于解决了，最后在大雕的提示下，开了路由SS全局代理，成功了，看样子前面是被墙了，再次感谢帮助我的人！

closed 2017-09-08 by coolsnowwolf

jackyan999, 2017-09-09:
谢谢你的回复，问题已经解决了，是因为网络被墙了引起的，后来改全局SS代理，一次通过编译成功，谢谢

MeISbug, 2017-09-09:
怎么还有联系方式，赶紧删了吧，不安全