coolsnowwolf/lede #29: 编译后ssr 守护日志提示错误

sirfromprc, 2017-09-09:
[2017-09-09 14:15:00] ShadowsocksR No Problem.
udhcpc: no lease, failing
udhcpc: sending discover
udhcpc: started, v1.27.2
/etc/rc.common: line 144: /usr/bin/ip: not found
/etc/rc.common: line 144: /usr/bin/ip: not found
 2017-09-09 14:00:09 INFO: obfs_param 
 2017-09-09 14:00:09 INFO: obfs plain
 2017-09-09 14:00:09 INFO: method chacha20-ietf
 2017-09-09 14:00:09 INFO: protocol_param 
 2017-09-09 14:00:09 INFO: protocol origin
/etc/rc.common: line 144: /usr/bin/ip: not found
/etc/rc.common: line 144: /usr/bin/ip: not found
udhcpc: no lease, failing
udhcpc: sending discover
udhcpc: started, v1.27.2
[2017-09-09 14:00:00] Problem decteted, restarting ShadowsocksR...

coolsnowwolf, 2017-09-09:
请将 default-settings 编译进去，或者参考

里面有个 ip-full 和 ip-tiny 的路径修正

coolsnowwolf, 2017-09-09:
算了，我来加吧。修复了。

你 git pull 一下再编译即可

sirfromprc, 2017-09-09:
还是不行，路由器里面/usr/bin/ip这个文件是链接到/usr/sbin/ip，但/usr/sbin/ip这个文件不存在。程序应该是执行的(opkg list-installed | grep ip-full >/dev/null 2>&1)  && ln -sf /usr/sbin/ip /usr/bin/ip这个命令。[ ! -f /usr/bin/ip ] && ln -sf /sbin/ip /usr/bin/ip这个命令没有执行，/sbin/ip 这个文件是存在的

coolsnowwolf, 2017-09-09:
看你选择了，你不要同时选择 ip-full 和 ip-tiny 即可

[image]

/usr/bin/ip 默认 在 ip-tiny 是真实文件

我也不知道官方为毛要这样分开

同样的问题出现在 MWAN3

coolsnowwolf, 2017-09-09:
提示：如果去掉绝对路径，在 LEDE 上面执行没问题，会自动判断路径，潘多拉上面有问题

coolsnowwolf, 2017-09-09:
ip-tiny 在这里 /usr/bin/ip
ip-full 在这里 /usr/sbin/ip

MWAN3 需要的是 /usr/bin/ip

coolsnowwolf, 2017-09-09:
另外 ， /sbin/ip 还有一个，是 busybox 生成的。。。。为毛这么绕，我也不明白

sirfromprc, 2017-09-09:
谢谢大雕，原来是只装了ip-full，程序应用没什么问题的，就是感觉变慢了。我又把ip-full和ip-tiny一起编译了，发现速度反而快了，明天再看看速度怎么样

closed 2017-09-11 by coolsnowwolf