coolsnowwolf/lede #44: SSR似乎不工作

MeISbug, 2017-09-13:
服务器是auth_aes128_md5+aes-192-ctr, 无法连上，无论是刷了固件的X86路由，还是WRT路由器，都一样。

用C#客户端，手机apk，都没问题。之前用的路由器固件，比如K2P的潘多拉，都没问题。

尝试切换到其它协议，也一样

coolsnowwolf, 2017-09-13:
工作的，K2P 潘多拉也是我做的

MeISbug, 2017-09-13:
是的，K2P用的就是你的固件，那个连接是正常的。

现在这个，在system log里看到:

Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available
Tue Sep 12 01:51:15 2017 kern.err /usr/bin/ssr-redir[9998]: accept: No file descriptors available

路由表：

Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         XX.XX.XX.XX      0.0.0.0         UG    0      0        0 pppoe-wan
10.0.1.0        *               255.255.255.0   U     0      0        0 br-lan
XX.XX.XX.XX     *               255.255.255.255 UH    0      0        0 pppoe-wan

只把Gateway下的IP隐去，其他信息真实

另外在/usr/bin里只找到ssr-redir, 没有local和tunnel，不知道是不是这个固件只用到redir?

MeISbug, 2017-09-13:
在/etc/下有shadowsocksr.json, 看过配置也是对的，是通过luci的那个应用配的

MeISbug, 2017-09-13:
运行iptables -t nat -L， 信息如下

Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
shadowsocksr_pre  tcp  --  anywhere             anywhere
ADBYBY     tcp  --  anywhere             anywhere             tcp dpt:www
REDIRECT   udp  --  anywhere             anywhere             udp dpt:domain redir ports 53
REDIRECT   tcp  --  anywhere             anywhere             tcp dpt:domain redir ports 53
prerouting_rule  all  --  anywhere             anywhere             /* !fw3: user chain for prerouting */
zone_lan_prerouting  all  --  anywhere             anywhere             /* !fw3 */
zone_wan_prerouting  all  --  anywhere             anywhere             /* !fw3 */
zone_wan_prerouting  all  --  anywhere             anywhere             /* !fw3 */

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
pdnsd_output  tcp  --  anywhere             anywhere             tcp dpt:domain
REDIRECT   tcp  --  anywhere             anywhere             multiport dports www,https ! match-set china dst redir ports 7070

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
postrouting_rule  all  --  anywhere             anywhere             /* !fw3: user chain for postrouting */
zone_lan_postrouting  all  --  anywhere             anywhere             /* !fw3 */
zone_wan_postrouting  all  --  anywhere             anywhere             /* !fw3 */
zone_wan_postrouting  all  --  anywhere             anywhere             /* !fw3 */

Chain ADBYBY (1 references)
target     prot opt source               destination
RETURN     all  --  anywhere             0.0.0.0/8
RETURN     all  --  anywhere             10.0.0.0/8
RETURN     all  --  anywhere             127.0.0.0/8
RETURN     all  --  anywhere             169.254.0.0/16
RETURN     all  --  anywhere             172.16.0.0/12
RETURN     all  --  anywhere             192.168.0.0/16
RETURN     all  --  anywhere             base-address.mcast.net/4
RETURN     all  --  anywhere             240.0.0.0/4
RETURN     all  --  anywhere             anywhere             match-set adbyby_esc dst
RETURN     all  --  anywhere             anywhere             ! match-set adbyby_wan dst
REDIRECT   tcp  --  anywhere             anywhere             redir ports 8118

Chain MINIUPNPD (1 references)
target     prot opt source               destination

Chain MINIUPNPD-POSTROUTING (1 references)
target     prot opt source               destination

Chain pdnsd_output (1 references)
target     prot opt source               destination
RETURN     all  --  anywhere             anywhere             match-set china dst
REDIRECT   tcp  --  anywhere             anywhere             redir ports 7070

Chain postrouting_lan_rule (1 references)
target     prot opt source               destination

Chain postrouting_rule (1 references)
target     prot opt source               destination

Chain postrouting_wan_rule (1 references)
target     prot opt source               destination

Chain prerouting_lan_rule (1 references)
target     prot opt source               destination

Chain prerouting_rule (1 references)
target     prot opt source               destination

Chain prerouting_wan_rule (1 references)
target     prot opt source               destination

Chain shadowsocksr_pre (1 references)
target     prot opt source               destination
RETURN     all  --  anywhere             anywhere             match-set local dst
RETURN     all  --  anywhere             xxx-xx-xxx-xx.vps.xxx.jp
RETURN     all  --  anywhere             anywhere             match-set china dst
REDIRECT   tcp  --  10.0.1.0/24          anywhere             redir ports 7070

Chain zone_lan_postrouting (1 references)
target     prot opt source               destination
postrouting_lan_rule  all  --  anywhere             anywhere             /* !fw3: user chain for postrouting */

Chain zone_lan_prerouting (1 references)
target     prot opt source               destination
prerouting_lan_rule  all  --  anywhere             anywhere             /* !fw3: user chain for prerouting */

Chain zone_wan_postrouting (2 references)
target     prot opt source               destination
MINIUPNPD-POSTROUTING  all  --  anywhere             anywhere
postrouting_wan_rule  all  --  anywhere             anywhere             /* !fw3: user chain for postrouting */
MASQUERADE  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_wan_prerouting (2 references)
target     prot opt source               destination
MINIUPNPD  all  --  anywhere             anywhere
prerouting_wan_rule  all  --  anywhere             anywhere             /* !fw3: user chain for prerouting */

coolsnowwolf, 2017-09-14:
你的 ssr 没有关联上 luci，请用默认脚本编译

MeISbug, 2017-09-14:
是按照说明编译的，没有什么修改

./scripts/feeds update -a 
 ./scripts/feeds install -a
 make menuconfig 

最后选好你要的路由，输入 make -j1 V=s 

唯一改的地方是把1改成5，因为我的是4核的电脑，想编译快一点。看到LEDE那边说是核心数+1。

这个应该不至于影响到SSR啊

coolsnowwolf, 2017-09-15:
在日志看，你没有 enable

coolsnowwolf, 2017-09-15:
配置没有写入，估计是你用了国产浏览器

MeISbug, 2017-09-15:
平时只用firefox的，国产的浏览器不敢用

ENABLED后在luci看到is running的提示，但是打不开google和国外网站，可能我截信息时关掉了。但是用scp连到路由器，看到配置是写了进去json文件的。

刚刚git pull了一下，重新编译，连luci都进不去了，但是能分配到IP

coolsnowwolf, 2017-09-16:
你编译的代码选的可能冲突了

closed 2017-09-17 by coolsnowwolf