coolsnowwolf/lede #46: amule编译报错

vincentchen, 2017-09-13:
不管装libwxbase2.8-dev libwxgtk2.8-dev还是libwxbase2.8-dbg libwxgtk2.8-dbg都不行
checking for File::Copy... ok
checking for the --with-toolkit option... will be automatically detected
checking for the --with-wxshared option... will be automatically detected
checking for the --with-wxdebug option... no
checking for the --with-wxversion option... will be automatically detected
checking for wx-config... /usr/bin/wx-config
checking for wxWidgets version >= 2.8.12 (--debug=no --unicode=yes)... no
configure: error: 
    The requested wxWidgets build couldn't be found.
    
    The configuration you asked for aMule requires a wxWidgets
    build with the following settings:
        --debug=no --unicode=yes
    but such build is not available.

    To see the wxWidgets builds available on this system, please use
    'wx-config --list' command. To use the default build, returned by
    'wx-config --selected-config', use the options with their 'auto'
    default values.

    If you still get this error, then check that 'wx-config' is
    in path, the directory where wxWidgets libraries are installed
    (returned by 'wx-config --libs' command) is in LD_LIBRARY_PATH
    or equivalent variable and wxWidgets version is 2.8.12 or above.
make[3]: *** [/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/aMule-2.3.2/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory `/root/lede/package/lean/amule'
make[2]: *** [package/lean/amule/compile] Error 2
make[2]: Leaving directory `/root/lede'
make[1]: *** [/root/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/root/lede'
make: *** [world] Error 2

coolsnowwolf, 2017-09-13:
请使用 ubuntu 14

vincentchen, 2017-09-13:
就是ubuntu14
Welcome to Ubuntu 14.04.5 LTS (GNU/Linux 3.13.0-32-generic x86_64)

 * Documentation:  https://help.ubuntu.com/

  System information as of Wed Sep 13 06:24:37 EDT 2017

  System load:  0.0                Processes:           92
  Usage of /:   85.9% of 18.04GB   Users logged in:     1
  Memory usage: 36%                IP address for eth0: 45.62.252.189
  Swap usage:   0%

  => / is using 85.9% of 18.04GB

  Graph this data and manage this system at:
    https://landscape.canonical.com/

New release '16.04.3 LTS' available.
Run 'do-release-upgrade' to upgrade to it.

coolsnowwolf, 2017-09-13:
wxWidgets 必须高于 2.8.12，请下载 wxWidgets 源代码编译好先

vincentchen, 2017-09-14:
还是不行，下了3.0.3
./configure --without-debug --enable-unicode
make 
make install 
报同样错误

coolsnowwolf, 2017-09-14:
如果编译的是 mips，那别想了

vincentchen, 2017-09-14:
K3

coolsnowwolf, 2017-09-15:
我测试了，一次过，和你的版本有关

vincentchen, 2017-09-15:
奇怪，我直接去build_dir/amule目录编译是不报错的

tomerilery, 2017-09-15:
ubuntu14.04LTS，wxWidget3.1版，不管有没有执行这条命令wx-congfig --debug=no --unicode=yes，编译报错：

checking for the --with-toolkit option... will be automatically detected
checking for the --with-wxshared option... will be automatically detected
checking for the --with-wxdebug option... no
checking for the --with-wxversion option... will be automatically detected
checking for wx-config... /usr/bin/wx-config
checking for wxWidgets version >= 2.8.12 (--debug=no --unicode=yes)... no
configure: error: 
    The requested wxWidgets build couldn't be found.
    
    The configuration you asked for aMule requires a wxWidgets
    build with the following settings:
        --debug=no --unicode=yes
    but such build is not available.

    To see the wxWidgets builds available on this system, please use
    'wx-config --list' command. To use the default build, returned by
    'wx-config --selected-config', use the options with their 'auto'
    default values.

    If you still get this error, then check that 'wx-config' is
    in path, the directory where wxWidgets libraries are installed
    (returned by 'wx-config --libs' command) is in LD_LIBRARY_PATH
    or equivalent variable and wxWidgets version is 2.8.12 or above.
make[3]: *** [/home/mltao/lede17.01/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/aMule-2.3.2/.configured_yyyyyyyn] Error 1
make[3]: Leaving directory `/home/mltao/lede17.01/source/package/lean/amule'
make[2]: *** [package/lean/amule/compile] Error 2
make[2]: Leaving directory `/home/mltao/lede17.01/source'
make[1]: *** [/home/mltao/lede17.01/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/mltao/lede17.01/source'
make: *** [world] 错误 2

coolsnowwolf, 2017-09-16:
我之前好像也遇过，忘了怎么解决了，大家试试再告诉我，我现在是可以成功编译的。。。代码没问题

2812015651, 2017-09-17:
缺了wxbase这个lib库

tomerilery, 2017-09-17:
sudo apt-get install wxbase* 后还是不行，编译通不过

已经安装了这个：
sudo apt-get install libwxbase3.1-0-unofficial libwxbase3.1-dev libwxgtk3.1-0-unofficial libwxgtk3.1-dev wx3.1-headers wx-common libwxgtk-webview3.1-0-unofficial libwxgtk-webview3.1-dev libwxgtk-media3.1-0-unofficial libwxgtk-media3.1-dev libwxbase3.1-0-unofficial-dbg libwxgtk3.1-0-unofficial-dbg libwxgtk-webview3.1-0-unofficial-dbg libwxgtk-media3.1-0-unofficial-dbg wx3.1-i18n wx3.1-examples

2812015651, 2017-09-17:
我说的是lede/op里的wxbase,看Makefile就知道了

tomerilery, 2017-09-17:
然而，在make menuconfig 里/ 搜索wxbase，提示selected by:package_amule [=y]

2812015651, 2017-09-18:
看下maz-1的

vincentchen, 2017-09-18:
楼上什么意思

lphgor, 2017-10-04:
项目源码里面没有wxbase，我自己补充以后，还是报错缺少crypto++。
```
checking for crypto++ version >= 5.1... no
configure: error: 
	Could not find crypto++ installation or sources.
	Please try again with --with-crypto-prefix=/my_crypto_prefix
	(replace /my_crypto_prefix with a valid path to your crypto++
	installation or source directory).
```
最后把 [https://github.com/openwrt-stuff/openwrt-extra/tree/master/libs](https://github.com/openwrt-stuff/openwrt-extra/tree/master/libs) 里面的 `libcryptopp` 和 `wxbase` 下载下来，重新编译就通过了。

dengxin6, 2017-10-05:
/package/libs/里缺了wxbase库 补上就好了

vincentchen, 2017-10-07:
楼上方法可行，编译通过

hzjnet, 2020-01-13:
不行，还是报错

alex9434, 2021-12-28:
I have exactly the same problem:
```

checking for wx-config... /home/alex/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/bin/wx-config
checking for wxWidgets version >= 2.8.12 (--debug=no --unicode=yes)... no
configure: error:
    The requested wxWidgets build couldn't be found.

    The configuration you asked for aMule requires a wxWidgets
    build with the following settings:
        --debug=no --unicode=yes
    but such build is not available.

    To see the wxWidgets builds available on this system, please use
    'wx-config --list' command. To use the default build, returned by
    'wx-config --selected-config', use the options with their 'auto'
    default values.

    If you still get this error, then check that 'wx-config' is
    in path, the directory where wxWidgets libraries are installed
    (returned by 'wx-config --libs' command) is in LD_LIBRARY_PATH
    or equivalent variable and wxWidgets version is 2.8.12 or above.
make[2]: *** [Makefile:115: /home/alex/openwrt/build_dir/target-aarch64_cortex-a72_musl/aMule-2.3.2/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
make[2]: Leaving directory '/home/alex/alex-openwrt/feeds/amule'
time: package/feeds/alexpackage/amule/compile#38.11#21.21#60.30
    ERROR: package/feeds/alexpackage/amule failed to build.
make[1]: *** [package/Makefile:114: package/feeds/alexpackage/amule/compile] Error 1
make[1]: Leaving directory '/home/alex/openwrt'
make: *** [/home/alex/openwrt/include/toplevel.mk:230: package/amule/compile] Error 2
```
Compiling on Debian Bullseye on arm64:
```
dpkg-query -l *wx*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version        Architecture Description
+++-======================-==============-============-===========================================================================
un  libwxbase3.0-0         <none>         <none>       (no description available)
ii  libwxbase3.0-0v5:arm64 3.0.5.1+dfsg-2 arm64        wxBase library (runtime) - non-GUI support classes of wxWidgets toolkit
ii  libwxbase3.0-dev:arm64 3.0.5.1+dfsg-2 arm64        wxBase library (development) - non-GUI support classes of wxWidgets toolkit
un  wx3.0-doc              <none>         <none>       (no description available)
ii  wx3.0-headers          3.0.5.1+dfsg-2 all          wxWidgets Cross-platform C++ GUI toolkit (header files)

```