coolsnowwolf/lede #47: 不知道什么地方编译错误啦。

caiyy, 2017-09-14:
CMakeFiles/blobmsg_json-static.dir/build.make:62: recipe for target 'CMakeFiles/blobmsg_json-static.dir/blobmsg_json.c.o' failed
make[6]: *** [CMakeFiles/blobmsg_json-static.dir/blobmsg_json.c.o] Error 1
make[6]: Leaving directory '/home/caiyy/桌面/lede/build_dir/target-arm_cortex-a9_musl_eabi/libubox-2017-06-17-fd57eea9'
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/blobmsg_json-static.dir/all' failed
make[5]: *** [CMakeFiles/blobmsg_json-static.dir/all] Error 2
make[5]: Leaving directory '/home/caiyy/桌面/lede/build_dir/target-arm_cortex-a9_musl_eabi/libubox-2017-06-17-fd57eea9'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/caiyy/桌面/lede/build_dir/target-arm_cortex-a9_musl_eabi/libubox-2017-06-17-fd57eea9'
Makefile:99: recipe for target '/home/caiyy/桌面/lede/build_dir/target-arm_cortex-a9_musl_eabi/libubox-2017-06-17-fd57eea9/.built' failed
make[3]: *** [/home/caiyy/桌面/lede/build_dir/target-arm_cortex-a9_musl_eabi/libubox-2017-06-17-fd57eea9/.built] Error 2
make[3]: Leaving directory '/home/caiyy/桌面/lede/package/libs/libubox'
package/Makefile:109: recipe for target 'package/libs/libubox/compile' failed
make[2]: *** [package/libs/libubox/compile] Error 2
make[2]: Leaving directory '/home/caiyy/桌面/lede'
package/Makefile:105: recipe for target '/home/caiyy/桌面/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/caiyy/桌面/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/caiyy/桌面/lede'
/home/caiyy/桌面/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2017-09-14:
没有翻墙导致的

coolsnowwolf, 2017-09-14:
还有，文件夹别用中文，有的你烦的

caiyy, 2017-09-14:
-。-！！我。显然是有翻墙呀。。

caiyy, 2017-09-14:
我再试试看。不用中文路径。

closed 2017-09-14 by coolsnowwolf

KuoKongQingYun, 2017-09-14:
遇到同样的问题，我是直接用国外服务器编译的，不存在翻墙的问题，路径也没中文
路由是极路由HC5962