coolsnowwolf/lede #55: youku-yk1 网口无法使用

huahuaxue, 2017-09-16:
今天编译优酷路由宝youku-yk1 黑色版，32M flash。发现无法网口无法使用，System log如下：

Thu Sep 14 22:56:51 2017 daemon.err uhttpd[1807]: Failed to connect to the switch. Use the "list" command to see which switches are available.
Thu Sep 14 22:56:56 2017 daemon.err uhttpd[1807]: Failed to connect to the switch. Use the "list" command to see which switches are available.
Thu Sep 14 22:57:01 2017 daemon.err uhttpd[1807]: Failed to connect to the switch. Use the "list" command to see which switches are available.
Thu Sep 14 23:00:17 2017 daemon.err uhttpd[1807]: Failed to connect to the switch. Use the "list" command to see which switches are available.
Thu Sep 14 23:00:22 2017 daemon.err uhttpd[1807]: Failed to connect to the switch. Use the "list" command to see which switches are available.
Thu Sep 14 23:00:27 2017 daemon.err uhttpd[1807]: Failed to connect to the switch. Use the "list" command to see which switches are available.

coolsnowwolf, 2017-09-16:
系统日志看看，虽然 mtk 不太懂

huahuaxue, 2017-09-17:
我搜索了这个问题貌似跟32m 闪存相关，应该是lede的问题。
我已经刷回pandorabox了。

closed 2017-09-17 by huahuaxue