coolsnowwolf/lede #56: Arch下编译报错

WarsFeng, 2017-09-16:
make[7]: Leaving directory '/home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4/man'
make[6]: Leaving directory '/home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4/man'
make[5]: Leaving directory '/home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4/man'
make[5]: Entering directory '/home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4'
make[6]: Entering directory '/home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4'
make[6]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/home/wars/Develop/Git/lede/staging_dir/host/lib/pkgconfig'
 /usr/bin/install -c -m 644 libcrypto.pc libssl.pc libtls.pc openssl.pc '/home/wars/Develop/Git/lede/staging_dir/host/lib/pkgconfig'
make[6]: Leaving directory '/home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4'
make[5]: Leaving directory '/home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4'
make[4]: Leaving directory '/home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4'
mkdir -p /home/wars/Develop/Git/lede/staging_dir/host/stamp
touch /home/wars/Develop/Git/lede/build_dir/host/libressl-2.5.4/.built
touch /home/wars/Develop/Git/lede/staging_dir/host/stamp/.libressl_installed
make[3]: Leaving directory '/home/wars/Develop/Git/lede/tools/libressl'
make[2]: Leaving directory '/home/wars/Develop/Git/lede'
make[1]: *** [tools/Makefile:146: /home/wars/Develop/Git/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynyynynnyyynny] Error 2
make[1]: Leaving directory '/home/wars/Develop/Git/lede'
make: *** [/home/wars/Develop/Git/lede/include/toplevel.mk:209：world] 错误 2

看了一下toplevel.mk:209是:
cp .config tmp/.config; 
-.-没毛病啊

Saimen-Luo, 2017-09-16:
还是老老实实按lean说的用Ubuntu吧，我之前用debian也是莫名其妙报错，用了Ubuntu就好了

coolsnowwolf, 2017-09-16:
用 Arch 要有自己解决深坑的能力，太激进了

closed 2017-09-16 by coolsnowwolf