coolsnowwolf/lede #58: ar71系列不会直接生成镜像，mtk系列没问题

ifwangs, 2017-09-17:
dd if=/media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/root.squashfs >> /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin
6608+1 records in
6608+1 records out
3383536 bytes (3.4 MB, 3.2 MiB) copied, 0.0158595 s, 213 MB/s
/media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/staging_dir/host/bin/mktplinkfw -H 0x09410005 -W 0x1 -F 4Mlzma -N OpenWrt -V r55-def8eaa -m 1 -k /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tl-wr941nd-v5-kernel.bin -r /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin -o /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin.new -j -X 0x40000 -a 0x4  -s && mv /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin.new /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin || rm -f /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin
[mktplinkfw] kernel length aligned to 1326196
[mktplinkfw] *** error: images are too big by 1040228 bytes
cp /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin /media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/bin/targets/ar71xx/generic/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin
cp: cannot stat '/media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/tmp/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin': No such file or directory
Makefile:64: recipe for target '/media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/bin/targets/ar71xx/generic/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin' failed
make[5]: [/media/wei/1962a254-5f71-48c6-a5a8-4596d7aa35d3/lede/bin/targets/ar71xx/generic/lede-ar71xx-generic-tl-wr941nd-v5-squashfs-sysupgrade.bin] Error 1 (ignored)

[image]

coolsnowwolf, 2017-09-17:
自己更改大小为 8MB

ifwangs, 2017-09-17:
明白了，谢谢

closed 2017-09-17 by ifwangs