coolsnowwolf/lede #59: 请教一下编译斐讯K3固件屏幕显示和adbyby plus问题

Saimen-Luo, 2017-09-17:
首先感谢lean大开源该代码，我编译了k3的固件，有两个问题解决不了。
一是屏幕无法正常显示，编译时选择了k3screenctrl 还是不行
[image]
[image]

二是adbyby plus规则一直无法更新，也过滤不了广告
[image]
[image]

还请不吝赐教！

tomerilery, 2017-09-17:
adbyby服务器被未知力量攻击了，请手动更新，ssh到路由，执行指令/usr/share/adbyby/adupdate.sh

Saimen-Luo, 2017-09-17:
@tomerilery 执行命令后规则更新了，adbyby是招谁惹谁了吗？还有你知道屏幕问题怎么解吗？现在是屏幕一直停留在开机界面，一直亮着。

gavin219, 2017-09-17:
因为没有配置k3screenctrl

Saimen-Luo, 2017-09-17:
可否细说一下配置方法，是这个luci吗？[https://github.com/Hill-98/luci-app-k3screenctrl](url) @gavin219

coolsnowwolf, 2017-09-17:
closed 的 issues 里面有写，认真看

Saimen-Luo, 2017-09-17:
OK，lean大最好在readme标明一下，像我一样不看closed 的 issues的人估计不少。

closed 2017-09-17 by Saimen-Luo