coolsnowwolf/lede #61: EA4500 单独编译zerotier ipk报错

qiankan520, 2017-09-18:
make[1]: Entering directory '/home/qian/lede'
make[2]: Entering directory '/home/qian/lede/package/libs/toolchain'
rm -rf /home/qian/lede/build_dir/target-arm_xscale_musl_eabi/toolchain/.pkgdir/libc.installed /home/qian/lede/build_dir/target-arm_xscale_musl_eabi/toolchain/.pkgdir/libc
mkdir -p /home/qian/lede/build_dir/target-arm_xscale_musl_eabi/toolchain/.pkgdir/libc
install -d -m0755 /home/qian/lede/build_dir/target-arm_xscale_musl_eabi/toolchain/.pkgdir/libc/lib /home/qian/lede/build_dir/target-arm_xscale_musl_eabi/toolchain/.pkgdir/libc/usr/bin
cp -fpR /home/qian/lede/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/lib/ld-musl-*.so* /home/qian/lede/build_dir/target-arm_xscale_musl_eabi/toolchain/.pkgdir/libc/lib/
cp: cannot stat '/home/qian/lede/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/lib/ld-musl-*.so*': No such file or directory
Makefile:618: recipe for target '/home/qian/lede/build_dir/target-arm_xscale_musl_eabi/toolchain/.pkgdir/libc.installed' failed
make[2]: *** [/home/qian/lede/build_dir/target-arm_xscale_musl_eabi/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/qian/lede/package/libs/toolchain'
package/Makefile:109: recipe for target 'package/libs/toolchain/compile' failed
make[1]: *** [package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/qian/lede'
/home/qian/lede/include/toplevel.mk:207: recipe for target 'package/zerotier/compile' failed
make: *** [package/zerotier/compile] Error 2

这是缺少什么库吗

coolsnowwolf, 2017-09-18:
缺少工具链，你没有编译过固件就直接编译软件是不可能的

qiankan520, 2017-09-18:
ok 明白了

closed 2017-09-19 by coolsnowwolf