coolsnowwolf/lede #63: 编译出错 求解

SUNJINZE, 2017-09-19:
mkdir -p /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/jz/lede/tmp/.root-copy.flock -c 'cp -fpR /home/jz/lede/build_dir/target-mipsel_24kc_musl/coreutils-8.23/.pkgdir/coreutils-nohup/. /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
make[8]: *** [base32.o] Error 1
make[8]: Leaving directory `/home/jz/lede/build_dir/target-mipsel_24kc_musl/wget-ssl/wget-1.19.1/lib'
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory `/home/jz/lede/build_dir/target-mipsel_24kc_musl/wget-ssl/wget-1.19.1/lib'
make[6]: *** [all] Error 2
make[6]: Leaving directory `/home/jz/lede/build_dir/target-mipsel_24kc_musl/wget-ssl/wget-1.19.1/lib'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/jz/lede/build_dir/target-mipsel_24kc_musl/wget-ssl/wget-1.19.1'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/jz/lede/build_dir/target-mipsel_24kc_musl/wget-ssl/wget-1.19.1'
make[3]: *** [/home/jz/lede/build_dir/target-mipsel_24kc_musl/wget-ssl/wget-1.19.1/.built] Error 2
make[3]: Leaving directory `/home/jz/lede/feeds/packages/net/wget'
make[2]: *** [package/feeds/packages/wget/compile] Error 2
make[2]: *** Waiting for unfinished jobs....
touch /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.coreutils-nohup_installed
echo "coreutils-nohup" >> /home/jz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/coreutils.default.install
make[3]: Leaving directory `/home/jz/lede/feeds/packages/utils/coreutils'
make[3]: Entering directory `/home/jz/lede/package/system/opkg'
make[3]: Leaving directory `/home/jz/lede/package/system/opkg'
make[3]: Entering directory `/home/jz/lede/package/libs/libubox'
tmp/.config-package.in:35181:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:35181:	symbol PACKAGE_luci-base is selected by PACKAGE_luci-mod-admin-full
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:35287:	symbol PACKAGE_luci-mod-admin-full is selected by PACKAGE_luci
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:35112:	symbol PACKAGE_luci is selected by PACKAGE_ipset-lists
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:79641:	symbol PACKAGE_ipset-lists is selected by PACKAGE_luci-app-shadowsocksr-pro
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:36430:	symbol PACKAGE_luci-app-shadowsocksr-pro depends on PACKAGE_iptables
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:55604:	symbol PACKAGE_iptables is selected by PACKAGE_sqm-scripts
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:963:	symbol PACKAGE_sqm-scripts is selected by PACKAGE_luci-app-sqm
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:36525:	symbol PACKAGE_luci-app-sqm depends on PACKAGE_luci-base
mkdir -p /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
echo '2017-06-17-fd57eea9' | cmp -s - /home/jz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.version || echo '2017-06-17-fd57eea9' > /home/jz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.version
SHELL= flock /home/jz/lede/tmp/.root-copy.flock -c 'cp -fpR /home/jz/lede/build_dir/target-mipsel_24kc_musl/libubox-2017-06-17-fd57eea9/.pkgdir/libubox/. /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libubox_installed
echo "libubox" >> /home/jz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.default.install
mkdir -p /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/jz/lede/tmp/.root-copy.flock -c 'cp -fpR /home/jz/lede/build_dir/target-mipsel_24kc_musl/libubox-2017-06-17-fd57eea9/.pkgdir/libblobmsg-json/. /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libblobmsg-json_installed
echo "libblobmsg-json" >> /home/jz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.default.install
mkdir -p /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/jz/lede/tmp/.root-copy.flock -c 'cp -fpR /home/jz/lede/build_dir/target-mipsel_24kc_musl/libubox-2017-06-17-fd57eea9/.pkgdir/jshn/. /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.jshn_installed
echo "jshn" >> /home/jz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.default.install
mkdir -p /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/jz/lede/tmp/.root-copy.flock -c 'cp -fpR /home/jz/lede/build_dir/target-mipsel_24kc_musl/libubox-2017-06-17-fd57eea9/.pkgdir/libjson-script/. /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libjson-script_installed
echo "libjson-script" >> /home/jz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libubox.default.install
make[3]: Leaving directory `/home/jz/lede/package/libs/libubox'
make[3]: Entering directory `/home/jz/lede/feeds/luci/modules/luci-base'
make[3]: Leaving directory `/home/jz/lede/feeds/luci/modules/luci-base'
make[2]: Leaving directory `/home/jz/lede'
make[3]: Entering directory `/home/jz/lede/package/firmware/linux-firmware'
mkdir -p /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/jz/lede/tmp/.root-copy.flock -c 'cp -fpR /home/jz/lede/build_dir/target-mipsel_24kc_musl/linux-firmware-2016-09-21-42ad5367/.pkgdir/rt2800-pci-firmware/. /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/jz/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.rt2800-pci-firmware_installed
echo "rt2800-pci-firmware" >> /home/jz/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory `/home/jz/lede/package/firmware/linux-firmware'
make[2]: Leaving directory `/home/jz/lede'
make[1]: *** [/home/jz/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/jz/lede'
Build failed - please re-run with -j1 to see the real error message
make: *** [world] 错误 1

coolsnowwolf, 2017-09-19:
下载问题，翻墙可破

SUNJINZE, 2017-09-19:
已经挂了全局，还是不行

closed 2017-09-22 by coolsnowwolf