coolsnowwolf/lede #65: wget-ssl 编译出错

vincentchen, 2017-09-19:
make[4]: Entering directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/wget-ssl/wget-1.19.1'
make  all-recursive
make[5]: Entering directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/wget-ssl/wget-1.19.1'
Making all in lib
make[6]: Entering directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/wget-ssl/wget-1.19.1/lib'
make  all-recursive
make[7]: Entering directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/wget-ssl/wget-1.19.1/lib'
make[8]: Entering directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/wget-ssl/wget-1.19.1/lib'
  CC       base32.o
In file included from /root/lede/staging_dir/toolchain-arm_cortex-a9_gcc-5.4.0_musl_eabi/include/endian.h:23:0,
                 from /root/lede/staging_dir/toolchain-arm_cortex-a9_gcc-5.4.0_musl_eabi/include/sys/types.h:70,
                 from ./sys/types.h:28,
                 from ./unistd.h:128,
                 from ./stdlib.h:96,
                 from base32.c:46:
./stdint.h:89:5: error: #if with no expression
make[8]: *** [base32.o] Error 1
make[8]: Leaving directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/wget-ssl/wget-1.19.1/lib'
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/wget-ssl/wget-1.19.1/lib'
make[6]: *** [all] Error 2
make[6]: Leaving directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/wget-ssl/wget-1.19.1/lib'
make[5]: *** [all-recursive] Error 1

coolsnowwolf, 2017-09-19:
工具链损坏了，clean一次后重新 make

vincentchen, 2017-09-19:
我把wget版本降回1.18，好了

tomerilery, 2017-09-20:
昨晚我也遇到这个问题了，make dirclean后再编译，还是出现这个错误，今早，我把wgwet换成1.18版本编译就通过了

closed 2017-09-29 by vincentchen