coolsnowwolf/lede #66: shadowsocksr-libev-gfwlist 编译这个什么错

vincentchen, 2017-09-19:
hadowsocksr-libev-gfwlist/CONTROL /root/lede/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo
install -d -m0755 /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/ipkg-arm_cortex-a9/shadowsocksr-libev-gfwlist/usr/bin
install -m0755 /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/src/ss-redir /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/ipkg-arm_cortex-a9/shadowsocksr-libev-gfwlist/usr/bin/ssr-redir
install -m0755 /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/src/ss-tunnel /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/ipkg-arm_cortex-a9/shadowsocksr-libev-gfwlist/usr/bin/ssr-tunnel
install: cannot stat '/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/src/ss-tunnel': No such file or directory
make[3]: *** [/root/lede/bin/packages/arm_cortex-a9/base/shadowsocksr-libev-gfwlist_2.5.1-1_arm_cortex-a9.ipk] Error 1
make[3]: Leaving directory `/root/lede/package/lean/shadowsocksR-libev-full'
make[2]: *** [package/lean/shadowsocksR-libev-full/compile] Error 2
make[2]: Leaving directory `/root/lede'

coolsnowwolf, 2017-09-19:
ss-tunnel 新的 ssr 代码中被删除了.... 我也要去除了

closed 2017-10-06 by coolsnowwolf

sdannyzfr, 2017-10-28:
install -m0755 /home/sdannyzfr/lede-17.01/build_dir/target-x86_64_musl-1.1.16/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/src/ss-tunnel /home/sdannyzfr/lede-17.01/staging_dir/target-x86_64_musl-1.1.16/root-x86/tmp-shadowsocksr-libev/usr/bin/ssr-tunnel
install: cannot stat '/home/sdannyzfr/lede-17.01/build_dir/target-x86_64_musl-1.1.16/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/src/ss-tunnel': No such file or directory
make[3]: *** [/home/sdannyzfr/lede-17.01/staging_dir/target-x86_64_musl-1.1.16/root-x86/stamp/.shadowsocksr-libev_installed] Error 1
make[3]: Leaving directory `/home/sdannyzfr/lede-17.01/package/lean/shadowsocksR-libev-full'
make[2]: *** [package/lean/shadowsocksR-libev-full/compile] Error 2
make[2]: Leaving directory `/home/sdannyzfr/lede-17.01'
make[1]: *** [/home/sdannyzfr/lede-17.01/staging_dir/target-x86_64_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/sdannyzfr/lede-17.01'
make: *** [world] 错误 2

请问怎么破？