coolsnowwolf/lede #70: 编译L大版本的固件，日志出现pdnsd缓存错误

tomerilery, 2017-09-20:
Wed Sep 20 09:00:24 2017 daemon.info pdnsd[25321]: pdnsd-1.2.9b-par starting.
Wed Sep 20 09:00:24 2017 daemon.warn pdnsd[25321]: Could not open disk cache file /var/pdnsd/pdnsd.cache: No such file or directory
Wed Sep 20 09:00:24 2017 daemon.warn pdnsd[25321]: Error: could not bind socket: No such file or directory.
Status readback will be impossible

怎么破？

SouHong, 2017-09-20:
已经关了CACHE

需要不显示这个
修改两处：

tomerilery, 2017-09-20:
楼上的，谢谢你的回复，希望你能说的更清楚一些，修改哪两处？

SouHong, 2017-09-20:
/etc/init.d/ssrpro
203行：perm_cache=off;  改为：perm_cache=512;

201 202行中间加入：
201--cat > /var/etc/pdnsd.conf <<EOF
+if ! test -f "/var/pdnsd/pdnsd.cache"; then
+dd if=/dev/zero of="/var/pdnsd/pdnsd.cache" bs=128KB count=4 2> /dev/null
+chown -R nobody.nogroup /var/pdnsd
+fi
202--global {
修改后还是提示其它。

daofeng2015, 2017-09-20:
没启用启动文件。。。。你没看makefile吗？

tomerilery, 2017-09-20:
@daofeng2015 去看makefile了，是这两行没有启用吗？
#$(INSTALL_DIR) $(1)/etc/init.d
#$(INSTALL_BIN) ./files/pdnsd.init $(1)/etc/init.d/pdnsd

daofeng2015, 2017-09-20:
前面二个#没看到吗？去掉试试

tomerilery, 2017-09-20:
@daofeng2015 你好，还有个问题想请教你，下面这个文件，你打开
https://github.com/coolsnowwolf/lede/blob/master/target/linux/mvebu/base-files/etc/init.d/set_cpu_affinity
例如这一段：
local irq_wifi0=`grep -E -m1 'mwlwifi' /proc/interrupts | cut -d: -f1 | tail -n1 | tr -d ' '`
	[ -n "$irq_wifi0" ] || echo "$name irq not found."
	echo "$val" > "/proc/irq/$irq_wifi0/smp_affinity"

grep -E -m1 'mwlwifi' /proc/interrupts | cut -d: -f1 | tail -n1 | tr -d ' '这个命令可以在路由器上执行并有返回值（返回值为44），[ -n "$irq_wifi0" ] 判断irq_wifi0的值是否为空，前面说的是这个值为44并不为空，但代码执行到这里却认为irq_wifi0的值为空 并且显示管道命令后的错误提示，导致$val的值不能写入smp_affinity，研究一天了，没搞明白，请赐教！

daofeng2015, 2017-09-21:
@tomerilery 没看过wifi这方面的代码

tomerilery, 2017-09-21:
@daofeng2015 @SouHong 你们两个说的方法不能去操作，不然ssr会工作不正常，大雕的默认状态就好，就是会提示磁盘缓存没有而已

coolsnowwolf, 2017-09-22:
已修复，其实不是问题，画蛇添足而已

closed 2017-09-22 by coolsnowwolf