coolsnowwolf/lede #71: （ARM）set_cpu_affinity日志出现错误

tomerilery, 2017-09-20:
机型是WRT1200AC，昨晚完整切换到L大固件，最新源码全新编译，使用后出现如下错误：
Tue Sep 19 23:36:55 2017 daemon.notice procd: /etc/rc.d/S99set_cpu_affinity: eth0 irq not found.
Tue Sep 19 23:36:55 2017 daemon.notice procd: /etc/rc.d/S99set_cpu_affinity: /etc/rc.common: line 361: can't create /proc/irq//smp_affinity: nonexistent directory
Tue Sep 19 23:36:55 2017 daemon.notice procd: /etc/rc.d/S99set_cpu_affinity: eth1 irq not found.
Tue Sep 19 23:36:55 2017 daemon.notice procd: /etc/rc.d/S99set_cpu_affinity: /etc/rc.common: line 361: can't create /proc/irq//smp_affinity: nonexistent directory

可以配合L大解决问题

vincentchen, 2017-09-20:
目录搞错了吧？/proc/irq//smp_affinity ==>/proc/irq/smp_affinity?试试

tomerilery, 2017-09-21:
@vincentchen 目录没有错，//中间本来是有内容的，但没有获取到值，所以空着。
下面这个文件，你打开
https://github.com/coolsnowwolf/lede/blob/master/target/linux/mvebu/base-files/etc/init.d/set_cpu_affinity
例如这一段：
local irq_wifi0=grep -E -m1 'mwlwifi' /proc/interrupts | cut -d: -f1 | tail -n1 | tr -d ' '
[ -n "$irq_wifi0" ] || echo "$name irq not found."
echo "$val" > "/proc/irq/$irq_wifi0/smp_affinity"

grep -E -m1 'mwlwifi' /proc/interrupts | cut -d: -f1 | tail -n1 | tr -d ' '这个命令可以在路由器上执行并有返回值（返回值为44），[ -n "$irq_wifi0" ] 判断irq_wifi0的值是否为空，前面说的是这个值为44并不为空，但代码执行到这里却认为irq_wifi0的值为空 并且显示管道命令后的错误提示，导致$val的值不能写入smp_affinity

closed 2017-09-26 by coolsnowwolf