coolsnowwolf/lede #78: openssl编译出错了

vincentchen, 2017-09-22:
make[5]: Entering directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/openssl-1.0.2l/crypto/engine'
../../util/domd ../.. -MD arm-openwrt-linux-muslgnueabi-gcc -- -fPIC -DOPENSSL_PIC -DZLIB_SHARED -DZLIB -DOPENSSL_THREADS -D_REENTRANT -DDSO_DLFCN -DHAVE_DLFCN_H -I/root/lede/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include -I/root/lede/staging_dir/target-arm_cortex-a9_musl_eabi/include -I/root/lede/staging_dir/toolchain-arm_cortex-a9_gcc-5.4.0_musl_eabi/usr/include -I/root/lede/staging_dir/toolchain-arm_cortex-a9_gcc-5.4.0_musl_eabi/include/fortify -I/root/lede/staging_dir/toolchain-arm_cortex-a9_gcc-5.4.0_musl_eabi/include -znow -zrelro -DOPENSSL_SMALL_FOOTPRINT -DHAVE_CRYPTODEV -DUSE_CRYPTODEV_DIGESTS -DOPENSSL_NO_ERR -DTERMIOS -Os -pipe -mcpu=cortex-a9 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/openssl-1.0.2l:openssl-1.0.2l -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -fomit-frame-pointer -Wall -I.. -I../.. -I../modes -I../asn1 -I../evp -I../../include  -DOPENSSL_NO_DEPRECATED -DOPENSSL_NO_CAMELLIA -DOPENSSL_NO_EC_NISTP_64_GCC_128 -DOPENSSL_NO_GMP -DOPENSSL_NO_IDEA -DOPENSSL_NO_JPAKE -DOPENSSL_NO_LIBUNBOUND -DOPENSSL_NO_MD2 -DOPENSSL_NO_MDC2 -DOPENSSL_NO_PERLASM -DOPENSSL_NO_RC5 -DOPENSSL_NO_RFC3779 -DOPENSSL_NO_SCTP -DOPENSSL_NO_SHA0 -DOPENSSL_NO_SSL_TRACE -DOPENSSL_NO_SSL2 -DOPENSSL_NO_SSL3 -DOPENSSL_NO_STORE -DOPENSSL_NO_UNIT_TEST -DOPENSSL_NO_WEAK_SSL_CIPHERS --  eng_err.c eng_lib.c eng_list.c eng_init.c eng_ctrl.c eng_table.c eng_pkey.c eng_fat.c eng_all.c tb_rsa.c tb_dsa.c tb_ecdsa.c tb_dh.c tb_ecdh.c tb_rand.c tb_store.c tb_cipher.c tb_digest.c tb_pkmeth.c tb_asnmth.c eng_openssl.c eng_cnf.c eng_dyn.c eng_cryptodev.c eng_rdrand.c
eng_cryptodev.c:57:31: fatal error: crypto/cryptodev.h: No such file or directory
compilation terminated.
make[5]: *** [depend] Error 1
make[5]: Leaving directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/openssl-1.0.2l/crypto/engine'
make[4]: *** [depend] Error 1
make[4]: Leaving directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/openssl-1.0.2l/crypto'
make[3]: *** [depend] Error 1
make[3]: Leaving directory `/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/openssl-1.0.2l'
make[2]: *** [/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/openssl-1.0.2l/.configured_75bf3d6cd39ab7c3e56e5dffee1fee84_shared_no-err_no-hw_zlib-dynamic_no-sse2_no-ssl2_-DHAVE_CRYPTODEV_-DUSE_CRYPTODEV_DIGESTS_no-ssl3_no-sse2_no-perlasm] Error 2
make[2]: Leaving directory `/root/lede/package/vincent/openssl'
make[1]: *** [package/vincent/openssl/compile] Error 2
make[1]: Leaving directory `/root/lede'

coolsnowwolf, 2017-09-26:
梯子问题

closed 2017-09-26 by coolsnowwolf

muziling, 2017-10-13:
https://github.com/openwrt/openwrt/tree/master/package/libs/openssl
https://github.com/openwrt-mirror/openwrt/tree/master/package/libs/openssl
这两个源刚好差了这个文件 crypto/cryptodev.h