coolsnowwolf/lede #79: DW33D没有5GHz无线

HiGarfield, 2017-09-23:
[   14.623653] PCI: Enabling device 0000:00:00.0 (0000 -> 0002)
[   14.629570] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   14.845292] ath10k_pci 0000:00:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:00:00.0.bin failed with error -2
[   14.856171] ath10k_pci 0000:00:00.0: Falling back to user helper
[   74.465578] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: 
[   74.862088] ath10k_pci 0000:00:00.0: Direct firmware load for ath10k/cal-pci-0000:00:00.0.bin failed with error -2
[   74.872615] ath10k_pci 0000:00:00.0: Falling back to user helper
[  106.190325] FAT-fs (sdb1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[  134.999641] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[  135.009043] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[  135.022080] ath10k_pci 0000:00:00.0: firmware ver 10.2.4-1.0-00029 api 5 features no-p2p,raw-mode,mfp,allows-mesh-bcast crc32 88595bb8
[  135.066989] ath10k_pci 0000:00:00.0: board id is not exist in otp, ignore it
[  135.074249] ath10k_pci 0000:00:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/board-2.bin failed with error -2
[  135.084851] ath10k_pci 0000:00:00.0: Falling back to user helper
[  144.692612] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store: map pages failed
[  144.712582] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[  144.836835] ath10k_pci 0000:00:00.0: otp calibration failed: 2
[  144.842775] ath10k_pci 0000:00:00.0: failed to run otp: -22
[  144.848426] ath10k_pci 0000:00:00.0: could not init core (-22)
[  144.854446] ath10k_pci 0000:00:00.0: could not probe fw (-22)

daofeng2015, 2017-09-23:
有无线啊。选择ath10k-firmware-qca988x就有5G了

HiGarfield, 2017-09-24:
现增加问题描述：
1、不添加软件包，编译后，5GHz正常。
2、添加N多软件包，编译成全家桶固件之后，启动奇慢。
3、编译出来的全家桶固件刷机完第一次启动，直接没5GHz，第二次（有时候包含第三次）重启，5GHz无线出来了，但无法正常工作。再次重启，5GHz无线正常，并且后面再重启，5GHz也正常。

我怀疑是全家桶固件启动过慢导致某些脚本没有成功执行。
不过代码我不懂，也只是猜测。

HiGarfield, 2017-09-24:
补充一些我全家桶固件添加的所有软件包：

root@LEDE:~# opkg list
adbyby - 2.7-20170823
aria2 - 1.32.0-3
ath10k-firmware-qca988x - 2017-03-29-956e2609-1
automount - 1-18
autosamba - 1-8
base-files - 174-r78-287575f
blkid - 2.30.1-2
block-mount - 2017-07-31-8ab4fda6-1
busybox - 1.27.2-1
coreutils - 8.23-3
coreutils-base64 - 8.23-3
coreutils-nohup - 8.23-3
ddns-scripts - 2.7.6-17
default-settings - 1.0-21
dnsmasq-full - 2.76-12
dosfstools - 4.1-1
dropbear - 2017.75-3
e2fsprogs - 1.43.5-1
etherwake - 1.09-4
fdisk - 2.30.1-2
firewall - 2017-05-27-a4d98aea-1
fstools - 2017-07-31-8ab4fda6-1
fwtool - 1
hd-idle - 1.04-2
hdparm - 9.50-1
hostapd-common - 2016-12-19-ad02e79d-4
ip-tiny - 4.11.0-1
ip6tables - 1.6.1-1
ipset - 6.32-2
ipset-lists - 20161112
iptables - 1.6.1-1
iptables-mod-conntrack-extra - 1.6.1-1
iptables-mod-ipopt - 1.6.1-1
iptables-mod-nat-extra - 1.6.1-1
iptables-mod-tproxy - 1.6.1-1
iw - 4.9-1
iwinfo - 2017-08-23-c1a03e82-1
jshn - 2017-06-17-fd57eea9-1
jsonfilter - 2016-07-02-dea067ad-1
kernel - 4.4.88-1-4290d981df12e76445f252412db00f26
kmod-ath - 4.4.88+2017-01-31-3
kmod-ath10k - 4.4.88+2017-01-31-3
kmod-ath9k - 4.4.88+2017-01-31-3
kmod-ath9k-common - 4.4.88+2017-01-31-3
kmod-cfg80211 - 4.4.88+2017-01-31-3
kmod-crypto-aead - 4.4.88-1
kmod-crypto-crc32c - 4.4.88-1
kmod-crypto-ecb - 4.4.88-1
kmod-crypto-hash - 4.4.88-1
kmod-crypto-manager - 4.4.88-1
kmod-crypto-null - 4.4.88-1
kmod-crypto-pcompress - 4.4.88-1
kmod-crypto-sha1 - 4.4.88-1
kmod-fast-classifier - 4.4.88-1
kmod-fs-exfat - 4.4.88+2017-01-03-8d291f525ce6d88fe0d8b11b86fd5c2e900401d3-1
kmod-fs-ext4 - 4.4.88-1
kmod-fs-vfat - 4.4.88-1
kmod-fuse - 4.4.88-1
kmod-gpio-button-hotplug - 4.4.88-2
kmod-gre - 4.4.88-1
kmod-ifb - 4.4.88-1
kmod-ip6tables - 4.4.88-1
kmod-ipt-conntrack - 4.4.88-1
kmod-ipt-conntrack-extra - 4.4.88-1
kmod-ipt-core - 4.4.88-1
kmod-ipt-ipopt - 4.4.88-1
kmod-ipt-ipset - 4.4.88-1
[... 173 lines cut ...]
swconfig - 11
tc - 4.11.0-1
terminfo - 6.0-1
transmission-daemon-mbedtls - 2.92+git-4
transmission-web - 2.92+git-4
ubi-utils - 1.5.2-1
uboot-envtools - 2015.10-1
ubox - 2017-09-01-b1bc8d5f-1
ubus - 2017-02-18-34c6e818-1
ubusd - 2017-02-18-34c6e818-1
uci - 2017-04-12-c4df32b3-1
uclient-fetch - 2017-09-06-24d6eded-1
uhttpd - 2016-10-25-1628fa4b-2
uhttpd-mod-ubus - 2016-10-25-1628fa4b-2
usign - 2015-07-04-ef641914-1
vlmcsd - svn1111-1
vsftpd-alt - 3.0.3-5
watchcat - 1-6
webui-aria2 - 2017-05-21-d1ce5b99-1
wget - 1.18-2
wifischedule - 1-2
wpad-mini - 2016-12-19-ad02e79d-4
xl2tpd - devel-20151125-3
yaaw - 2017-04-11-1
zlib - 1.2.11-1

coolsnowwolf, 2017-09-26:
这个嘛。。。等我买到 DW33D 再看看了

HiGarfield, 2017-09-28:
第一次启动，/etc/config/wireless 压根就没5GHz。多重启几遍才有。有了之后就没事了。以后不会丢了。

closed 2017-10-01 by coolsnowwolf