coolsnowwolf/lede #94: luci-app-sqm recursive dependency

waterun, 2017-09-30:
luci-app-sqm 递归依赖，导致选择菜单里经常看不到该安装选项：
tmp/.config-package.in:35702:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:35702:	symbol PACKAGE_luci-base is selected by PACKAGE_luci-mod-admin-full
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:35808:	symbol PACKAGE_luci-mod-admin-full is selected by PACKAGE_luci
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:35633:	symbol PACKAGE_luci is selected by PACKAGE_ipset-lists
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:80251:	symbol PACKAGE_ipset-lists is selected by PACKAGE_luci-app-shadowsocksr-pro
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:36982:	symbol PACKAGE_luci-app-shadowsocksr-pro depends on PACKAGE_iptables
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:56206:	symbol PACKAGE_iptables is selected by PACKAGE_sqm-scripts
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:963:	symbol PACKAGE_sqm-scripts is selected by PACKAGE_luci-app-sqm
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:37077:	symbol PACKAGE_luci-app-sqm depends on PACKAGE_luci-base

coolsnowwolf, 2017-10-01:
官方的 feed sqm 问题，请到官方提交

closed 2017-10-01 by coolsnowwolf

uglee, 2018-03-17:
删除/mnt/sdb/lede-v17.01/package/lean/ipset-lists/Makefile里面+luci
这个属于重复引用导致报错。