coolsnowwolf/lede #100: 编译无线不能用请教，但是官网用官网15051和17012 无线都能正常配置使用！

hzjnet, 2017-10-03:
硬件是mr3420 v1 ，官网固件无线可使用的配置文件为：
cat /etc/config/wireless
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'pci0000:00/0000:00:00.0'
        option htmode 'HT20'
        option country 'US'
config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option encryption 'none'
        option ssid '2214LEDE'

使用这个配置文件我编译的固件还是不能使用（mode始终是unkown），但是页面配置里面的scan又能扫描出来周边的AP信号！

coolsnowwolf, 2017-10-03:
没有系统日志，不评论

K2P 没有开源的驱动

hzjnet, 2017-10-03:
先请教 mr3420的吧，贴上log，可惜日志为空！  不知道找对没有：
root@LEDE:~# find /var/log |xargs ls -la
-rw-r--r--    1 root     root           745 Oct  4 07:09 /var/log/ddns/myddns_ipv4.log
-rw-r--r--    1 root     root           853 Oct  4 07:09 /var/log/ddns/myddns_ipv6.log
-rw-r--r--    1 root     root             0 Oct  4 06:00 /var/log/lastlog
-rw-r--r--    1 root     root             0 Oct  4 06:00 /var/log/wtmp

/var/log:
drwxr-xr-x    3 root     root           100 Oct  4 07:09 .
drwxrwxrwt   20 root     root           540 Oct  4 07:10 ..
drwxr-xr-x    2 root     root            80 Oct  4 07:09 ddns
-rw-r--r--    1 root     root             0 Oct  4 06:00 lastlog
-rw-r--r--    1 root     root             0 Oct  4 06:00 wtmp

/var/log/ddns:
drwxr-xr-x    2 root     root            80 Oct  4 07:09 .
drwxr-xr-x    3 root     root           100 Oct  4 07:09 ..
-rw-r--r--    1 root     root           745 Oct  4 07:09 myddns_ipv4.log
-rw-r--r--    1 root     root           853 Oct  4 07:09 myddns_ipv6.log

coolsnowwolf, 2017-10-14:
当然不能用，目前没有开源的驱动是支持 K2P 无线的

closed 2017-10-14 by coolsnowwolf