coolsnowwolf/lede #105: R7800 刷固件2小时后无2.4Gwifi 附上系统日志

ClayMoreBoy, 2017-10-05:
Thu Oct  5 13:41:06 2017 daemon.info hostapd: wlan1: STA f0:b4:29:98:06:de IEEE 802.11: authenticated
Thu Oct  5 13:41:06 2017 daemon.info hostapd: wlan1: STA f0:b4:29:98:06:de IEEE 802.11: associated (aid 1)
Thu Oct  5 13:41:07 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED f0:b4:29:98:06:de
Thu Oct  5 13:41:07 2017 daemon.info hostapd: wlan1: STA f0:b4:29:98:06:de WPA: pairwise key handshake completed (RSN)
Thu Oct  5 13:41:07 2017 daemon.info pdnsd[5275]: pdnsd-1.2.9b-par starting.
Thu Oct  5 13:41:07 2017 daemon.warn pdnsd[5275]: Cache file /var/pdnsd/pdnsd.cache ignored because of incompatible version identifier
Thu Oct  5 13:41:07 2017 user.notice ddns-scripts[5322]: myddns_ipv4: PID '5322' started at 2017-10-05 13:41
Thu Oct  5 13:41:07 2017 user.warn ddns-scripts[5322]: myddns_ipv4: Service section disabled! - TERMINATE
Thu Oct  5 13:41:07 2017 user.warn ddns-scripts[5322]: myddns_ipv4: PID '5322' exit WITH ERROR '1' at 2017-10-05 13:41
Thu Oct  5 13:41:08 2017 kern.err /usr/bin/ssr-redir[1172]: failed to resolve server name, wait 8 seconds
Thu Oct  5 13:41:08 2017 kern.info /usr/bin/ssr-redir[1172]: initializing ciphers... chacha20
Thu Oct  5 13:41:08 2017 kern.info /usr/bin/ssr-redir[1172]: tcp port reuse enabled
Thu Oct  5 13:41:08 2017 kern.info /usr/bin/ssr-redir[1172]: UDP relay enabled
Thu Oct  5 13:41:08 2017 kern.info /usr/bin/ssr-redir[1172]: udp port reuse enabled
Thu Oct  5 13:41:08 2017 kern.info /usr/bin/ssr-redir[1172]: listening at 0.0.0.0:7070
Thu Oct  5 13:41:08 2017 kern.info /usr/bin/ssr-redir[1172]: running from root user
Thu Oct  5 13:41:08 2017 daemon.info hostapd: wlan0: STA 20:82:c0:d7:3c:a7 IEEE 802.11: authenticated
Thu Oct  5 13:41:08 2017 daemon.info hostapd: wlan0: STA 20:82:c0:d7:3c:a7 IEEE 802.11: associated (aid 1)
Thu Oct  5 13:41:08 2017 daemon.notice hostapd: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 20:82:c0:d7:3c:a7
Thu Oct  5 13:41:09 2017 daemon.info hostapd: wlan1: STA 48:db:50:80:83:79 IEEE 802.11: authenticated
Thu Oct  5 13:41:09 2017 daemon.info hostapd: wlan1: STA 48:db:50:80:83:79 IEEE 802.11: associated (aid 2)
Thu Oct  5 13:41:09 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 48:db:50:80:83:79
Thu Oct  5 13:41:09 2017 daemon.info hostapd: wlan1: STA 48:db:50:80:83:79 WPA: pairwise key handshake completed (RSN)
Thu Oct  5 13:41:09 2017 daemon.notice hostapd: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 20:82:c0:d7:3c:a7
Thu Oct  5 13:41:10 2017 daemon.info dnsmasq-dhcp[4348]: DHCPDISCOVER(br-lan) f0:b4:29:98:06:de 
Thu Oct  5 13:41:10 2017 daemon.info dnsmasq-dhcp[4348]: DHCPOFFER(br-lan) 192.168.1.249 f0:b4:29:98:06:de 
Thu Oct  5 13:41:10 2017 daemon.info dnsmasq-dhcp[4348]: DHCPDISCOVER(br-lan) f0:b4:29:98:06:de 
Thu Oct  5 13:41:10 2017 daemon.info dnsmasq-dhcp[4348]: DHCPOFFER(br-lan) 192.168.1.249 f0:b4:29:98:06:de 
Thu Oct  5 13:41:10 2017 daemon.info dnsmasq-dhcp[4348]: DHCPREQUEST(br-lan) 192.168.1.189 48:db:50:80:83:79 
Thu Oct  5 13:41:10 2017 daemon.info dnsmasq-dhcp[4348]: DHCPACK(br-lan) 192.168.1.189 48:db:50:80:83:79 HUAWEI_RIO-CL00-526d6bcc9
Thu Oct  5 13:41:10 2017 daemon.info dnsmasq-dhcp[4348]: DHCPREQUEST(br-lan) 192.168.1.249 f0:b4:29:98:06:de 
Thu Oct  5 13:41:10 2017 daemon.info dnsmasq-dhcp[4348]: DHCPACK(br-lan) 192.168.1.249 f0:b4:29:98:06:de zhimi-airpurifier-m1_miio44516046
Thu Oct  5 13:41:10 2017 daemon.notice hostapd: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 20:82:c0:d7:3c:a7
Thu Oct  5 13:41:11 2017 daemon.info hostapd: wlan0: STA 7c:01:91:b2:15:c6 IEEE 802.11: authenticated
Thu Oct  5 13:41:11 2017 daemon.info hostapd: wlan0: STA 7c:01:91:b2:15:c6 IEEE 802.11: associated (aid 2)
Thu Oct  5 13:41:11 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 7c:01:91:b2:15:c6
Thu Oct  5 13:41:11 2017 daemon.info hostapd: wlan0: STA 7c:01:91:b2:15:c6 WPA: pairwise key handshake completed (RSN)
Thu Oct  5 13:41:11 2017 daemon.info dnsmasq-dhcp[4348]: DHCPREQUEST(br-lan) 192.168.1.103 7c:01:91:b2:15:c6 
Thu Oct  5 13:41:11 2017 daemon.info dnsmasq-dhcp[4348]: DHCPACK(br-lan) 192.168.1.103 7c:01:91:b2:15:c6 iPhone
Thu Oct  5 13:41:11 2017 daemon.notice hostapd: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 20:82:c0:d7:3c:a7
Thu Oct  5 13:41:12 2017 daemon.warn odhcpd[1766]: DHCPV6 SOLICIT IA_NA from 0001000120ce7f4a94de8008c6cd on br-lan: ok fd9f:a6f7:4cd8::86b/128 
Thu Oct  5 13:41:12 2017 daemon.warn odhcpd[1766]: DHCPV6 REQUEST IA_NA from 0001000120ce7f4a94de8008c6cd on br-lan: ok fd9f:a6f7:4cd8::86b/128 
Thu Oct  5 13:41:12 2017 daemon.info dnsmasq[4348]: read /etc/hosts - 4 addresses
Thu Oct  5 13:41:12 2017 daemon.info dnsmasq[4348]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Thu Oct  5 13:41:12 2017 daemon.info dnsmasq-dhcp[4348]: read /etc/ethers - 0 addresses
Thu Oct  5 13:41:12 2017 daemon.info dnsmasq[4348]: read /etc/hosts - 4 addresses
Thu Oct  5 13:41:12 2017 daemon.info dnsmasq[4348]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Thu Oct  5 13:41:12 2017 daemon.info dnsmasq-dhcp[4348]: read /etc/ethers - 0 addresses
Thu Oct  5 13:41:17 2017 daemon.warn odhcpd[1766]: DHCPV6 CONFIRM IA_NA from 0001000120ce7f4a94de8008c6cd on br-lan: not on-link fd9f:a6f7:4cd8::86b/128 
Thu Oct  5 13:41:17 2017 daemon.warn odhcpd[1766]: DHCPV6 SOLICIT IA_NA from 0001000120ce7f4a94de8008c6cd on br-lan: ok fd9f:a6f7:4cd8::86b/128 
Thu Oct  5 13:41:17 2017 daemon.info dnsmasq[4348]: read /etc/hosts - 4 addresses
Thu Oct  5 13:41:17 2017 daemon.info dnsmasq[4348]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Thu Oct  5 13:41:17 2017 daemon.info dnsmasq-dhcp[4348]: read /etc/ethers - 0 addresses
Thu Oct  5 13:41:17 2017 daemon.info hostapd: wlan0: STA 20:82:c0:d7:3c:a7 IEEE 802.11: deauthenticated due to local deauth request
Thu Oct  5 13:41:18 2017 daemon.warn odhcpd[1766]: DHCPV6 REQUEST IA_NA from 0001000120ce7f4a94de8008c6cd on br-lan: ok fd9f:a6f7:4cd8::86b/128 
Thu Oct  5 13:41:18 2017 daemon.info dnsmasq[4348]: read /etc/hosts - 4 addresses
Thu Oct  5 13:41:18 2017 daemon.info dnsmasq[4348]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Thu Oct  5 13:41:18 2017 daemon.info dnsmasq-dhcp[4348]: read /etc/ethers - 0 addresses
Thu Oct  5 13:41:22 2017 daemon.info hostapd: wlan1: STA 20:82:c0:d7:3c:a7 IEEE 802.11: authenticated
Thu Oct  5 13:41:22 2017 daemon.info hostapd: wlan1: STA 20:82:c0:d7:3c:a7 IEEE 802.11: associated (aid 3)
Thu Oct  5 13:41:22 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 20:82:c0:d7:3c:a7
Thu Oct  5 13:41:22 2017 daemon.info hostapd: wlan1: STA 20:82:c0:d7:3c:a7 WPA: pairwise key handshake completed (RSN)
Thu Oct  5 13:41:25 2017 daemon.info dnsmasq-dhcp[4348]: DHCPDISCOVER(br-lan) 20:82:c0:d7:3c:a7 
Thu Oct  5 13:41:25 2017 daemon.info dnsmasq-dhcp[4348]: DHCPOFFER(br-lan) 192.168.1.140 20:82:c0:d7:3c:a7 
Thu Oct  5 13:41:25 2017 daemon.info dnsmasq-dhcp[4348]: DHCPREQUEST(br-lan) 192.168.1.140 20:82:c0:d7:3c:a7 
Thu Oct  5 13:41:25 2017 daemon.info dnsmasq-dhcp[4348]: DHCPACK(br-lan) 192.168.1.140 20:82:c0:d7:3c:a7 Mi-4c-xiaomishouji
Thu Oct  5 13:41:36 2017 daemon.info dnsmasq[4348]: read /etc/hosts - 4 addresses
Thu Oct  5 13:41:36 2017 daemon.info dnsmasq[4348]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Thu Oct  5 13:41:36 2017 daemon.info dnsmasq-dhcp[4348]: read /etc/ethers - 0 addresses
Thu Oct  5 13:47:57 2017 daemon.notice netifd: Interface 'wan_6' is now up
[... 128 lines cut ...]
Thu Oct  5 15:42:15 2017 daemon.warn odhcpd[1766]: A default route is present but there is no public prefix on br-lan thus we don't announce a default route!
Thu Oct  5 15:49:51 2017 daemon.warn odhcpd[1766]: A default route is present but there is no public prefix on br-lan thus we don't announce a default route!
Thu Oct  5 15:51:01 2017 daemon.info hostapd: wlan1: STA f0:b4:29:98:06:de WPA: group key handshake completed (RSN)
Thu Oct  5 15:51:01 2017 daemon.info hostapd: wlan1: STA 20:82:c0:d7:3c:a7 WPA: group key handshake completed (RSN)
Thu Oct  5 15:51:01 2017 daemon.info hostapd: wlan0: STA 7c:01:91:b2:15:c6 WPA: group key handshake completed (RSN)
Thu Oct  5 15:53:43 2017 daemon.warn odhcpd[1766]: A default route is present but there is no public prefix on br-lan thus we don't announce a default route!
Thu Oct  5 15:57:01 2017 daemon.info hostapd: wlan1: STA dc:2b:2a:a8:db:d8 IEEE 802.11: authenticated
Thu Oct  5 15:57:01 2017 daemon.info hostapd: wlan1: STA dc:2b:2a:a8:db:d8 IEEE 802.11: associated (aid 2)
Thu Oct  5 15:57:01 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:2b:2a:a8:db:d8
[... cut ...]
Thu Oct  5 16:01:04 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 20:82:c0:d7:3c:a7
Thu Oct  5 16:01:04 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED f0:b4:29:98:06:de
Thu Oct  5 16:01:09 2017 daemon.info hostapd: wlan1: STA dc:2b:2a:a8:db:d8 IEEE 802.11: deauthenticated due to local deauth request
Thu Oct  5 16:01:09 2017 daemon.info hostapd: wlan1: STA 20:82:c0:d7:3c:a7 IEEE 802.11: deauthenticated due to local deauth request
Thu Oct  5 16:01:09 2017 daemon.info hostapd: wlan1: STA f0:b4:29:98:06:de IEEE 802.11: deauthenticated due to local deauth request

qyh214, 2017-10-11:
建议你重刷一次，如有必要进入救砖模式，然后重刷，用最新的

coolsnowwolf, 2017-10-14:
trunk 版本问题，用新版的 ath10k-ct 驱动和 fw

closed 2017-10-14 by coolsnowwolf