coolsnowwolf/lede #112: HC5962.dts  + usb2.0端口

SouHong, 2017-10-09:
/dts-v1/;

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
	compatible = "hiwifi,hc5962", "mediatek,mt7621-soc";
	model = "HiWiFi HC5962";

	memory@0 {
		device_type = "memory";
		reg = <0x0 0x10000000>;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	gpio-leds {
		compatible = "gpio-leds";

		status {
			label = "hc5962:white:status";
			gpios = <&gpio0 6 GPIO_ACTIVE_LOW>;
		};

		system {
			label = "hc5962:red:system";
			gpios = <&gpio0 7 GPIO_ACTIVE_LOW>;
		};
	};

	gpio_export {
		compatible = "gpio-export";
		#size-cells = <0>;

		usbpower {
			gpio-export,name = "usbpower";
			gpio-export,output = <1>;
			gpios = <&gpio0 12 0>;
		};
	};

	gpio-keys-polled {
		compatible = "gpio-keys-polled";
		#address-cells = <1>;
		#size-cells = <0>;
		poll-interval = <20>;

		reset {
			label = "reset";
			gpios = <&gpio0 18 GPIO_ACTIVE_HIGH>;
			linux,code = <KEY_RESTART>;
		};
	};
};

&nand {
	status = "okay";

	partition@0 {
		label = "u-boot";
		reg = <0x0 0x80000>;
		read-only;
	};

	partition@80000 {
		label = "debug";
		reg = <0x80000 0x80000>;
		read-only;
	};

	factory: partition@100000 {
		label = "factory";
		reg = <0x100000 0x40000>;
		read-only;
	};

	partition@140000 {
		label = "kernel";
		reg = <0x140000 0x200000>;
	};

	partition@340000 {
		label = "ubi";
		reg = <0x340000 0x1E00000>;
	};

	partition@2140000 {
		label = "hw_panic";
		reg = <0x2140000 0x80000>;
		read-only;
	};

	partition@21c0000 {
		label = "bdinfo";
		reg = <0x21c0000 0x80000>;
		read-only;
	};

	partition@2240000 {
		label = "backup";
		reg = <0x2240000 0x80000>;
		read-only;
	};

	partition@22c0000 {
		label = "overly";
		reg = <0x22c0000 0x1000000>;
	};

	partition@32c0000 {
		label = "firmware_backup";
		reg = <0x32c0000 0x2000000>;
	};

	partition@52c0000 {
		label = "oem";
		reg = <0x52c0000 0x200000>;
	};

	partition@54c0000 {
		label = "opt";
		reg = <0x54c0000 0x2ac0000>;
	};
};

&pcie {
	status = "okay";

	pcie0 {
		mt76@0,0 {
			reg = <0x0000 0 0 0 0>;
			device_type = "pci";
			mediatek,mtd-eeprom = <&factory 0x0000>;
			ieee80211-freq-limit = <2400000 2500000>;
		};
	};

	pcie1 {
		mt76@1,0 {
			reg = <0x0000 0 0 0 0>;
			device_type = "pci";
			mediatek,mtd-eeprom = <&factory 0x8000>;
			ieee80211-freq-limit = <5000000 6000000>;
		};
	};
};

&pinctrl {
	state_default: pinctrl0 {
		gpio {
			ralink,group = "uart3", "jtag";
			ralink,function = "gpio";
		};
	};
};

coolsnowwolf, 2017-10-12:
请 pull 相关代码，我来合并

Ray-CHN, 2017-10-12:
想帮你 但是要建分支 太烦了

SouHong, 2017-10-13:
@coolsnowwolf 已经pull

closed 2017-10-14 by SouHong