coolsnowwolf/lede #125: brcmfmac4366c-pcie.bin 是原版固件？

jackwish, 2017-10-19:
看到 https://github.com/coolsnowwolf/lede/blob/master/package/lean/k3-brcmfmac4366c-firmware/files/lib/firmware/brcm/brcmfmac4366c-pcie.bin 比较疑惑

coolsnowwolf, 2017-10-20:
不是

closed 2017-10-20 by coolsnowwolf

adaaaaaa, 2020-02-22:
@coolsnowwolf TP的WDN8280网卡也是这个固件。。。
好像驱动不起来。。。

[    7.312368] Broadcom 43xx driver loaded [ Features: PNL ]
[    7.313664] usbcore: registered new interface driver brcmfmac
[    7.314243] brcmfmac 0000:01:00.0: enabling device (0000 -> 0002)
[    7.435217] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4366c-pcie for chip BCM4366/4
[    7.436097] brcmfmac 0000:01:00.0: Direct firmware load for brcm/brcmfmac4366c-pcie.bin failed with error -2
[    7.437028] brcmfmac 0000:01:00.0: Falling back to syfs fallback for: brcm/brcmfmac4366c-pcie.bin
[    7.441526] firmware brcm!brcmfmac4366c-pcie.bin: firmware_loading_store: map pages failed