coolsnowwolf/lede #131: 编译flex时出现Segmentation fault

dsanke, 2017-10-20:
编译lede官方源码没问题，看了下也是同样的flex2.6.4
`dsanke@Ubuntu:~/lede$ make V=s
make[1]: Entering directory '/home/dsanke/lede'
make[2]: Entering directory '/home/dsanke/lede'
+ mkdir -p /home/dsanke/lede/staging_dir/target-mips_24kc_musl
+ cd /home/dsanke/lede/staging_dir/target-mips_24kc_musl
+ mkdir -p bin lib include stamp
mkdir -p /home/dsanke/lede/build_dir/target-mips_24kc_musl/stamp
touch /home/dsanke/lede/staging_dir/target-mips_24kc_musl/.prepared
+ mkdir -p /home/dsanke/lede/staging_dir/host
+ cd /home/dsanke/lede/staging_dir/host
+ mkdir -p bin lib include stamp
mkdir -p /home/dsanke/lede/build_dir/host/stamp /home/dsanke/lede/staging_dir/host/include/sys
install -m0644 /home/dsanke/lede/tools/include/*.h /home/dsanke/lede/staging_dir/host/include/
install -m0644 /home/dsanke/lede/tools/include/sys/*.h /home/dsanke/lede/staging_dir/host/include/sys/
ln -sf lib /home/dsanke/lede/staging_dir/host/lib64
touch /home/dsanke/lede/staging_dir/host/.prepared
make[3]: Entering directory '/home/dsanke/lede/tools/flock'
make[3]: Leaving directory '/home/dsanke/lede/tools/flock'
make[3]: Entering directory '/home/dsanke/lede/tools/sed'
make[3]: Leaving directory '/home/dsanke/lede/tools/sed'
make[3]: Entering directory '/home/dsanke/lede/tools/patch'
make[3]: Leaving directory '/home/dsanke/lede/tools/patch'
make[3]: Entering directory '/home/dsanke/lede/tools/xz'
make[3]: Leaving directory '/home/dsanke/lede/tools/xz'
make[3]: Entering directory '/home/dsanke/lede/tools/tar'
make[3]: Leaving directory '/home/dsanke/lede/tools/tar'
make[3]: Entering directory '/home/dsanke/lede/tools/m4'
make[3]: Leaving directory '/home/dsanke/lede/tools/m4'
make[3]: Entering directory '/home/dsanke/lede/tools/autoconf'
make[3]: Leaving directory '/home/dsanke/lede/tools/autoconf'
make[3]: Entering directory '/home/dsanke/lede/tools/pkg-config'
make[3]: Leaving directory '/home/dsanke/lede/tools/pkg-config'
make[3]: Entering directory '/home/dsanke/lede/tools/automake'
make[3]: Leaving directory '/home/dsanke/lede/tools/automake'
make[3]: Entering directory '/home/dsanke/lede/tools/missing-macros'
make[3]: Leaving directory '/home/dsanke/lede/tools/missing-macros'
make[3]: Entering directory '/home/dsanke/lede/tools/libtool'
make[3]: Leaving directory '/home/dsanke/lede/tools/libtool'
make[3]: Entering directory '/home/dsanke/lede/tools/gmp'
make[3]: Leaving directory '/home/dsanke/lede/tools/gmp'
make[3]: Entering directory '/home/dsanke/lede/tools/mpfr'
make[3]: Leaving directory '/home/dsanke/lede/tools/mpfr'
make[3]: Entering directory '/home/dsanke/lede/tools/mpc'
make[3]: Leaving directory '/home/dsanke/lede/tools/mpc'
make[3]: Entering directory '/home/dsanke/lede/tools/libelf'
make[3]: Leaving directory '/home/dsanke/lede/tools/libelf'
make[3]: Entering directory '/home/dsanke/lede/tools/expat'
make[3]: Leaving directory '/home/dsanke/lede/tools/expat'
make[3]: Entering directory '/home/dsanke/lede/tools/flex'
CFLAGS="-O2 -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include " CPPFLAGS="-I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include " CXXFLAGS="" LDFLAGS="-L/home/dsanke/lede/staging_dir/host/lib -L/home/dsanke/lede/staging_dir/host/usr/lib " make  -C /home/dsanke/lede/build_dir/host/flex-2.6.4  
make[4]: Entering directory '/home/dsanke/lede/build_dir/host/flex-2.6.4'
Making all in src
make[5]: Entering directory '/home/dsanke/lede/build_dir/host/flex-2.6.4/src'
make  all-am
make[6]: Entering directory '/home/dsanke/lede/build_dir/host/flex-2.6.4/src'
depbase=`echo libmain.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/usr/bin/env bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.  -DLOCALEDIR=\"/home/dsanke/lede/staging_dir/host/share/locale\" -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include   -O2 -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include  -MT libmain.lo -MD -MP -MF $depbase.Tpo -c -o libmain.lo libmain.c &&\
mv -f $depbase.Tpo $depbase.Plo
OpenWrt-libtool: compile:  gcc -DHAVE_CONFIG_H -I. -DLOCALEDIR=\"/home/dsanke/lede/staging_dir/host/share/locale\" -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include -O2 -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include -MT libmain.lo -MD -MP -MF .deps/libmain.Tpo -c libmain.c -o libmain.o
/usr/bin/env bash ../libtool  --tag=CC   --mode=link gcc  -O2 -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include  -version-info 2:0:0 -L/home/dsanke/lede/staging_dir/host/lib -L/home/dsanke/lede/staging_dir/host/usr/lib  -o libfl.la -rpath /home/dsanke/lede/staging_dir/host/lib libmain.lo libyywrap.lo  -lm 
OpenWrt-libtool: link: rm -fr  .libs/libfl.a .libs/libfl.la .libs/libfl.lai
OpenWrt-libtool: link: ar cru .libs/libfl.a  libmain.o libyywrap.o
ar: `u' modifier ignored since `D' is the default (see `U')
OpenWrt-libtool: link: ranlib .libs/libfl.a
OpenWrt-libtool: link: ( cd ".libs" && rm -f "libfl.la" && ln -s "../libfl.la" "libfl.la" )
gcc -DHAVE_CONFIG_H -I.  -DLOCALEDIR=\"/home/dsanke/lede/staging_dir/host/share/locale\" -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include    -O2 -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include  -MT flex-buf.o -MD -MP -MF .deps/flex-buf.Tpo -c -o flex-buf.o `test -f 'buf.c' || echo './'`buf.c
mv -f .deps/flex-buf.Tpo .deps/flex-buf.Po
gcc -DHAVE_CONFIG_H -I.  -DLOCALEDIR=\"/home/dsanke/lede/staging_dir/host/share/locale\" -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include    -O2 -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include  -MT flex-ccl.o -MD -MP -MF .deps/flex-ccl.Tpo -c -o flex-ccl.o `test -f 'ccl.c' || echo './'`ccl.c
mv -f .deps/flex-ccl.Tpo .deps/flex-ccl.Po
[... 101 lines cut ...]
mv -f .deps/stage1flex-yylex.Tpo .deps/stage1flex-yylex.Po
/usr/bin/env bash ../libtool  --tag=CC   --mode=link gcc  -O2 -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include   -L/home/dsanke/lede/staging_dir/host/lib -L/home/dsanke/lede/staging_dir/host/usr/lib  -o stage1flex stage1flex-scan.o stage1flex-buf.o stage1flex-ccl.o stage1flex-dfa.o stage1flex-ecs.o stage1flex-filter.o stage1flex-gen.o stage1flex-main.o stage1flex-misc.o stage1flex-nfa.o stage1flex-options.o stage1flex-parse.o stage1flex-regex.o stage1flex-scanflags.o stage1flex-scanopt.o stage1flex-skel.o stage1flex-sym.o stage1flex-tables.o stage1flex-tables_shared.o stage1flex-tblcmp.o stage1flex-yylex.o    -lm 
OpenWrt-libtool: link: gcc -O2 -I/home/dsanke/lede/staging_dir/host/include -I/home/dsanke/lede/staging_dir/host/usr/include -o stage1flex stage1flex-scan.o stage1flex-buf.o stage1flex-ccl.o stage1flex-dfa.o stage1flex-ecs.o stage1flex-filter.o stage1flex-gen.o stage1flex-main.o stage1flex-misc.o stage1flex-nfa.o stage1flex-options.o stage1flex-parse.o stage1flex-regex.o stage1flex-scanflags.o stage1flex-scanopt.o stage1flex-skel.o stage1flex-sym.o stage1flex-tables.o stage1flex-tables_shared.o stage1flex-tblcmp.o stage1flex-yylex.o  -L/home/dsanke/lede/staging_dir/host/lib -L/home/dsanke/lede/staging_dir/host/usr/lib -lm
./stage1flex   -o stage1scan.c ./scan.l
Makefile:1689: recipe for target 'stage1scan.c' failed
make[6]: *** [stage1scan.c] Segmentation fault (core dumped)
make[6]: Leaving directory '/home/dsanke/lede/build_dir/host/flex-2.6.4/src'
Makefile:539: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/dsanke/lede/build_dir/host/flex-2.6.4/src'
Makefile:525: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/dsanke/lede/build_dir/host/flex-2.6.4'
Makefile:28: recipe for target '/home/dsanke/lede/build_dir/host/flex-2.6.4/.built' failed
make[3]: *** [/home/dsanke/lede/build_dir/host/flex-2.6.4/.built] Error 2
make[3]: Leaving directory '/home/dsanke/lede/tools/flex'
tools/Makefile:150: recipe for target 'tools/flex/compile' failed
make[2]: *** [tools/flex/compile] Error 2
make[2]: Leaving directory '/home/dsanke/lede'
tools/Makefile:148: recipe for target '/home/dsanke/lede/staging_dir/target-mips_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyyyyyyyyyyyyyyyyyynnyynynyyyynny' failed
make[1]: *** [/home/dsanke/lede/staging_dir/target-mips_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyyyyyyyyyyyyyyyyyynnyynynyyyynny] Error 2
make[1]: Leaving directory '/home/dsanke/lede'
/home/dsanke/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
`

dsanke, 2017-10-20:
发现lede官方有修复：https://github.com/lede-project/source/commit/0fb14a2b1ab2f82ce63f4437b062229d73d90516

closed 2017-10-20 by dsanke

arfaWong, 2017-10-21:
@dsanke 请问这个要怎么修复？

dsanke, 2017-10-21:
@arfaWong 下载 https://raw.githubusercontent.com/lede-project/source/master/tools/flex/patches/200-build-AC_USE_SYSTEM_EXTENSIONS-in-configure.ac.patch
放到 源码/tools/flex/patches里面

chd1998, 2018-01-03:
要用那个110开头的patch...