coolsnowwolf/lede #135: 编译报错

arfaWong, 2017-10-21:
```
stage1flex-gen.o stage1flex-main.o stage1flex-misc.o stage1flex-nfa.o stage1flex-options.o stage1flex-parse.o stage1flex-regex.o stage1flex-scanflags.o stage1flex-scanopt.o stage1flex-skel.o stage1flex-sym.o stage1flex-tables.o stage1flex-tables_shared.o stage1flex-tblcmp.o stage1flex-yylex.o  -L/home/saro/lede/staging_dir/host/lib -L/home/saro/lede/staging_dir/host/usr/lib -lm
./stage1flex   -o stage1scan.c ./scan.l
Makefile:1689: recipe for target 'stage1scan.c' failed
make[6]: *** [stage1scan.c] Segmentation fault (core dumped)
make[6]: Leaving directory '/home/saro/lede/build_dir/host/flex-2.6.4/src'
Makefile:539: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/saro/lede/build_dir/host/flex-2.6.4/src'
Makefile:525: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/saro/lede/build_dir/host/flex-2.6.4'
Makefile:28: recipe for target '/home/saro/lede/build_dir/host/flex-2.6.4/.built' failed
make[3]: *** [/home/saro/lede/build_dir/host/flex-2.6.4/.built] Error 2
make[3]: Leaving directory '/home/saro/lede/tools/flex'
tools/Makefile:150: recipe for target 'tools/flex/compile' failed
make[2]: *** [tools/flex/compile] Error 2
make[2]: Leaving directory '/home/saro/lede'
tools/Makefile:148: recipe for target '/home/saro/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyyynynnyyynny' failed
make[1]: *** [/home/saro/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyyynynnyyynny] Error 2
make[1]: Leaving directory '/home/saro/lede'
/home/saro/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
```

hxy8241, 2017-10-27:
同样的错误，应该是新系统什么东西升级了，我昨天用linuxmint还能编译成功的。今天装了ubuntu17.10就失败了。

HuLian21, 2017-10-28:
我在进行 ./scripts/feeds install -a时出现了这一问题
WARNING: No feed for package 'libsqlite2' found, maybe it's already part of the standard packages?
WARNING: No feed for package 'flite' found, maybe it's already part of the standard packages?
WARNING: No feed for package 'libyuv' found, maybe it's already part of the standard packages?
WARNING: No feed for package 'libilbc' found, maybe it's already part of the standard packages?
WARNING: No feed for package 'libmemcached' found, maybe it's already part of the standard packages?
WARNING: No feed for package 'libsilk' found, maybe it's already part of the standard packages?
WARNING: No feed for package 'libg7221' found, maybe it's already part of the standard packages?
WARNING: No feed for package 'freeradius-client' found, maybe it's already part of the standard packages?
通过SUDO安装libssh-dev 也不行 貌似是包缺失，会有影响吗

hxy8241, 2017-10-28:
这个我编译成功时也出现的，应该没关系的，好像是进入标准包库了。

arfaWong, 2017-11-03:
#131

closed 2017-11-03 by arfaWong