coolsnowwolf/lede #151: 编译k2正常，但是编译x86_64则一直报错。过不了这个lm-sensors，又不知道如何取消它

hcg101, 2017-10-28:
/home/hcg/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
prog/sensors/Module.mk:43: recipe for target 'prog/sensors/sensors' failed
make[4]: *** [prog/sensors/sensors] Error 1
make[4]: Leaving directory '/home/hcg/lede/build_dir/target-x86_64_musl/lm-sensors-3.3.5'
Makefile:116: recipe for target '/home/hcg/lede/build_dir/target-x86_64_musl/lm-sensors-3.3.5/.built' failed
make[3]: *** [/home/hcg/lede/build_dir/target-x86_64_musl/lm-sensors-3.3.5/.built] Error 2
make[3]: Leaving directory '/home/hcg/lede/feeds/packages/utils/lm-sensors'
package/Makefile:109: recipe for target 'package/feeds/packages/lm-sensors/compile' failed
make[2]: *** [package/feeds/packages/lm-sensors/compile] Error 2
make[2]: Leaving directory '/home/hcg/lede'
package/Makefile:105: recipe for target '/home/hcg/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/hcg/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hcg/lede'
/home/hcg/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2

daofeng2015, 2017-10-28:
musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status

HuLian21, 2017-10-29:
make[3]: *** [/home/lede/lede/build_dir/target-mipsel_24kc_musl/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory `/home/lede/lede/package/lean/shadowsocksR-libev-full'
make[2]: *** [package/lean/shadowsocksR-libev-full/compile] Error 2
make[2]: Leaving directory `/home/lede/lede'
make[1]: *** [/home/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/lede/lede'
make: *** [world] Error 2
也出现问题了 我编译的是NEWIFI Y1

closed 2018-10-06 by coolsnowwolf