coolsnowwolf/lede #166: acrh17分区数据

wisdomjxc, 2017-11-04:
下载地址：http://pan.baidu.com/s/1nvR3ix7

dev:    size   erasesize  name
mtd0: 00400000 00020000 "Bootloader"
mtd1: 07c00000 00020000 "UBI_DEV"（备份失败，太大，路由放不下）
mtd2: 0005d000 0001f000 "nvram"
mtd3: 0001f000 0001f000 "Factory"
mtd4: 0001f000 0001f000 "Factory2"
mtd5: 03013000 0001f000 "linux"
mtd6: 02e13eac 0001f000 "rootfs"
mtd7: 03013000 0001f000 "linux2"
mtd8: 02e13eac 0001f000 "rootfs2"

ptpt52, 2018-01-02:
哥们，这个设备能支持吗？

wisdomjxc, 2018-01-03:
@ptpt52 现在无线还不行，最近大雕忙于其他，只能等了。

ptpt52, 2018-01-03:
@wisdomjxc 我看了，并非无线不行，而是 flash 不能被识别

wisdomjxc, 2018-01-03:
@ptpt52 分区相对于AC58U，有所改动，还是按照AC58U做的，还没有修改。

ptpt52, 2018-01-03:
@wisdomjxc flash 驱动有问题，或者硬件不被识别

wisdomjxc, 2018-01-03:
@ptpt52 分区地址问题，没有修正，驱动是没问题的，高通毕竟开源的，这款路由器的原厂固件的代码也开源了

ptpt52, 2018-01-03:
你看启动记录
`
kern.err kernel: [ 0.984992] m25p80 spi0.0: unrecognized JEDEC id bytes: ff, ff, ff`

ptpt52, 2018-01-03:
没有正确识别 flash

ptpt52, 2018-01-03:
```
Sat Nov 4 22:22:56 2017 kern.err kernel: [ 41.654143] mt29f spi0.1: WAIT timedout!!!
Sat Nov 4 22:22:56 2017 kern.err kernel: [ 81.674142] mt29f spi0.1: wait timedout!
```

ptpt52, 2018-01-03:
@wisdomjxc 原厂固件代码路径发我一下，我看看

wisdomjxc, 2018-01-03:
@ptpt52 https://www.asus.com.cn/Networking/RT-ACRH17/HelpDesk_Download/   操作系统选其他即可看到

ptpt52, 2018-01-04:
@wisdomjxc @coolsnowwolf 
我现在让 设备跑起来了，2.4G可以，但是5G不行
LED灯， LAN1-LAN4灯，状态灯可以，
wan灯不行，其他也不行，我不知道wan灯的GPIO

https://github.com/ptpt52/openwrt-openwrt

swshan, 2018-01-23:
在lede论坛看到楼上的发言 哈哈

closed 2018-10-06 by coolsnowwolf