coolsnowwolf/lede #167: mount_root: failed to mount -t jffs2 /dev/mtdblock8 /tmp/overlay: I/O error

aooom, 2017-11-05:
Sun Nov  5 12:35:58 2017 kern.notice kernel: [    0.000000] Linux version 4.9.60 (aabb@aboo-vm) (gcc version 5.4.0 (LEDE GCC 5.4.0 r78-287575f) ) #0 Sun Nov 5 04:35:33 2017
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    0.000000] MyLoader: sysp=00000000, boardp=00000000, parts=00000000
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] SoC: Atheros AR7161 rev 2
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Determined physical RAM map:
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000]  memory: 08000000 @ 00000000 (usable)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Nov  5 12:35:58 2017 kern.warn kernel: [    0.000000] OF: fdt:No valid device tree found, continuing without
Sun Nov  5 12:35:58 2017 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Sun Nov  5 12:35:58 2017 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Zone ranges:
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Early memory node ranges
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 804ad834, node_mem_map 81000020
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    0.000000]   Normal zone: 256 pages used for memmap
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
Sun Nov  5 12:35:58 2017 kern.notice kernel: [    0.000000] Kernel command line:  board=WZR-HP-AG300H console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Memory: 124552K/131072K available (3396K kernel code, 168K rwdata, 844K rodata, 320K init, 214K bss, 6520K reserved, 0K cma-reserved)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] NR_IRQS:51
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] Clocks: CPU:680.000MHz, DDR:340.000MHz, AHB:170.000MHz, Ref:40.000MHz
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5621354254 ns
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.000008] sched_clock: 32 bits at 340MHz, resolution 2ns, wraps every 6316128254ns
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.007798] Calibrating delay loop... 452.19 BogoMIPS (lpj=2260992)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.070514] pid_max: default: 32768 minimum: 301
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.075219] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.081828] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.090885] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.100743] futex hash table entries: 256 (order: -1, 3072 bytes)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.107305] NET: Registered protocol family 16
Sun Nov  5 12:35:58 2017 kern.info kernel: [    0.112901] MIPS: machine is Buffalo WZR-HP-AG300H/WZR-600DHP
Sun Nov  5 12:35:58 2017 kern.notice kernel: [    2.572902] random: fast init done
Sun Nov  5 12:35:58 2017 kern.warn kernel: [    2.795568] registering PCI controller with io_map_base unset
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.812122] PCI host bridge to bus 0000:00
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.816277] pci_bus 0000:00: root bus resource [mem 0x10000000-0x16ffffff]
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.823131] pci_bus 0000:00: root bus resource [io  0x0000]
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.828700] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.835472] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    2.843424] pci 0000:00:11.0: [168c:ff1d] type 00 class 0x020000
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.843434] pci 0000:00:11.0: fixup device configuration
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    2.849953] pci 0000:00:11.0: reg 0x10: [mem 0x00000000-0x0000ffff]
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    2.850007] pci 0000:00:11.0: PME# supported from D0 D3hot
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    2.850191] pci 0000:00:12.0: [168c:ff1d] type 00 class 0x020000
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.850200] pci 0000:00:12.0: fixup device configuration
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    2.856733] pci 0000:00:12.0: reg 0x10: [mem 0x00000000-0x0000ffff]
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    2.856781] pci 0000:00:12.0: PME# supported from D0 D3hot
Sun Nov  5 12:35:58 2017 kern.debug kernel: [    2.856953] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.856979] pci 0000:00:11.0: BAR 0: assigned [mem 0x10000000-0x1000ffff]
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.863755] pci 0000:00:12.0: BAR 0: assigned [mem 0x10010000-0x1001ffff]
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.870563] pci 0000:00:11.0: using irq 40 for pin 1
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.875516] pci 0000:00:12.0: using irq 41 for pin 1
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.881066] clocksource: Switched to clocksource MIPS
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.886957] NET: Registered protocol family 2
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.892150] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.899122] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
Sun Nov  5 12:35:58 2017 kern.info kernel: [    2.905519] TCP: Hash tables configured (established 1024 bind 1024)
[... 399 lines cut ...]
Sun Nov  5 21:41:54 2017 daemon.info dnsmasq[4509]: using local addresses only for domain example.com
Sun Nov  5 21:41:54 2017 daemon.info dnsmasq[4509]: using nameserver 114.114.114.114#53 for domain valf.atm.youku.com 
Sun Nov  5 21:41:54 2017 daemon.info dnsmasq[4509]: using 3 more local addresses
Sun Nov  5 21:41:54 2017 daemon.info dnsmasq[4509]: reading /tmp/resolv.conf.auto
Sun Nov  5 21:41:54 2017 daemon.info dnsmasq[4509]: using local addresses only for domain test
Sun Nov  5 21:41:54 2017 daemon.info dnsmasq[4509]: using local addresses only for domain onion
Sun Nov  5 21:41:54 2017 daemon.info dnsmasq[4509]: using local addresses only for domain localhost
[... cut ...]
Sun Nov  5 21:41:55 2017 daemon.notice procd: /etc/rc.d/S99ssrpro: WARNING: Shadowsocksr is disabled.
Sun Nov  5 21:41:55 2017 user.notice ddns-scripts[4511]: myddns_ipv4: PID '4511' started at 2017-11-05 21:41
Sun Nov  5 21:41:55 2017 user.warn ddns-scripts[4511]: myddns_ipv4: Service section disabled! - TERMINATE
Sun Nov  5 21:41:55 2017 daemon.info procd: - init complete -
Sun Nov  5 21:41:55 2017 user.warn ddns-scripts[4511]: myddns_ipv4: PID '4511' exit WITH ERROR '1' at 2017-11-05 21:41
Sun Nov  5 21:41:56 2017 daemon.info urandom_seed[4614]: Seed saved (/etc/urandom.seed)

aooom, 2017-11-05:
blk_update_request: I/O error,也很多看不懂

flipphos, 2017-12-12:
我编译生成的vmdk镜像启动也出现同样的
blk_update_request: I/O error
错误。

closed 2018-10-06 by coolsnowwolf