coolsnowwolf/lede #182: 编译Wndr3800固件报错。中途退出

lemonict, 2017-11-15:
make[7]: *** No rule to make target 'arch/mips/ath79/mach-tl-wr941n-v7.o', needed by 'arch/mips/ath79/built-in.o'.  Stop.
scripts/Makefile.build:544: recipe for target 'arch/mips/ath79' failed
make[6]: *** [arch/mips/ath79] Error 2
Makefile:991: recipe for target 'arch/mips' failed
make[5]: *** [arch/mips] Error 2
make[5]: Leaving directory '/home/am2dk/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.61'
Makefile:24: recipe for target '/home/am2dk/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.61/.image' failed
make[4]: *** [/home/am2dk/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.61/.image] Error 2
make[4]: Leaving directory '/home/am2dk/lede/target/linux/ar71xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/am2dk/lede/target/linux'
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/am2dk/lede'
target/Makefile:19: recipe for target '/home/am2dk/lede/staging_dir/target-mips_24kc_musl/stamp/.target_install' failed
make[1]: *** [/home/am2dk/lede/staging_dir/target-mips_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/am2dk/lede'
/home/am2dk/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2

kingyan, 2017-11-15:
缺文件，把这个文件加上就行了。
https://github.com/coolsnowwolf/lede/pull/179/commits/9893339684445b7870c67f3b4b72654cd589235d

lemonict, 2017-11-15:
能具体点儿吗，放到哪个目录里面？恳请指导

kingyan, 2017-11-15:
target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wr941n-v7.c

lemonict, 2017-11-15:
小白问题，没看到.c文件的下载链接～～

kingyan, 2017-11-15:
https://github.com/coolsnowwolf/lede/commit/9893339684445b7870c67f3b4b72654cd589235d
这个就是文件链接啊，你点开就是了。

lemonict, 2017-11-15:
点开是一个网页，如何直接下载到这个.c文件？

q158073378252010, 2017-11-24:
@lemonict 
 点开[9893339](https://github.com/coolsnowwolf/lede/commit/9893339684445b7870c67f3b4b72654cd589235d)--->>> CTRL+F 查找 “view” 并点击  --->>> CTRL+F 查找 “Raw” 并点击

lemonict, 2017-11-25:
非常感谢！！

closed 2018-10-06 by coolsnowwolf