coolsnowwolf/lede #191: wrt3200运行一段时间后5G连不上，系统无响应，请L查看一下日志

gavin219, 2017-11-18:
[sys.txt](https://github.com/coolsnowwolf/lede/files/1484416/sys.txt)
[kern.txt](https://github.com/coolsnowwolf/lede/files/1484417/kern.txt)

fuqiang03, 2017-11-20:
这个错误是wrt的通病，只能换固件。

gavin219, 2017-11-21:
@fuqiang03 换什么固件呢？

fuqiang03, 2017-11-21:
@gavin219 换其他固件就可以了，这个错误是编译时候产生的，我在编译的时候也遇到了这个错误，但是没有得到本质的解决，只是想办法避免了而已

gavin219, 2017-11-22:
@fuqiang03 通过什么方法避免的？官方固件也有这个问题吗？

kmahyyg, 2017-11-22:
我的k2p也有这个问题

fuqiang03, 2017-11-22:
@gavin219 @kmahyyg 这个问题是在构建固件的时候产生的（不排除部分插件的错误或者虚拟机下编译也可能出现这样错误），当问题发生时，一定要清楚所有编译的缓存，重新配置编译的配置文件。一般就可以解决了。
我没有k2p机器所以我不了解。

kmahyyg, 2017-11-22:
@fuqiang03 应该是编译配置的问题，还请大佬明示。能提供下您的配置文件吗？

fuqiang03, 2017-11-22:
[source4.4.tar.gz](https://github.com/coolsnowwolf/lede/files/1496820/source4.4.tar.gz)

gavin219, 2017-11-23:
@fuqiang03 我全新git的。编译过程没有出现错误。现在没有编译任何插件。再试试。不知道官方的有没有问题？

kmahyyg, 2017-11-23:
这个问题我翻了下历史issue，感觉是openwrt系的通病，应该是什么上游代码有问题。还请开发者检查修正。    @coolsnowwolf

gavin219, 2017-11-24:
@fuqiang03 
@kmahyyg 
我临时写了个计划任务，每隔6个小时重启一下无线。目前运行了20多个小时还没出问题

kmahyyg, 2017-11-24:
我的基本一大流量之后3h左右就挂，太烦了。不想折腾。没流量挂一天都没事.......

fuqiang03, 2017-11-24:
如果其他机出现了，同样的错误代码，就可以好判断了，如果是刷的是他人的固件，只能换，如果是自己编译的，清除编译缓存，包括，package目录下feed的链接，重新配置，一般都能解决问题，目前怀疑，mac80211,hostapd.因为报错的代码可以在里面找到。一有大流量数据，无线就宕机，所以弄的我每次编固都要试下。不过，这个问题出现的概率不高。

closed 2018-10-06 by coolsnowwolf