coolsnowwolf/lede #200: 无法编译

evindu, 2017-11-25:
`checking whether mknod can create fifo without root privileges... configure: error: in `/root/lede/build_dir/host/tar-1.29':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[3]: *** [/root/lede/build_dir/host/tar-1.29/.configured] Error 1
make[3]: Leaving directory `/root/lede/tools/tar'
make[2]: *** [tools/tar/compile] Error 2
make[2]: Leaving directory `/root/lede'
make[1]: *** [/root/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynnyynynnyyynny] Error 2
make[1]: Leaving directory `/root/lede'
make: *** [world] Error 2
root@ubuntu:~/lede#`

ctw8989, 2017-11-26:
configure: error: you should not run configure as root

cuijin1985, 2017-11-26:
新建一个具有sudo权限的用户，然后再编译，root会报错。

evindu, 2017-11-26:
解决了 直接 export FORCE_UNSAFE_CONFIGURE=1，谢谢解答

closed 2018-10-06 by coolsnowwolf