coolsnowwolf/lede #207: Netgear R7800编译到usbip出错，hash值计算不一致。

badsirius, 2017-11-29:
编译错误提示：

>  make[3]: Entering directory `/home/lede/lede/feeds/packages/net/usbip'
> mkdir -p /home/lede/lede/dl
> SHELL= flock /home/lede/lede/tmp/.usb.ids.flock -c '  	/home/lede/lede/scripts/download.pl "/home/lede/lede/dl" "usb.ids" "47dc941dca801b89f1529e09f0d338755d8fbb0ce75f526e1dcccc4fbf8fc66c" "" "http://www.linux-usb.org/"    '
> + curl -f --connect-timeout 20 --retry 5 --location --insecure http://www.linux-usb.org/usb.ids
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100  578k  100  578k    0     0   171k      0  0:00:03  0:00:03 --:--:--  171k
> Hash of the downloaded file does not match (file: 85029230a978a62b2cc8e81d0c9fa3d18eaa881e52483e55eeab2024caf041de, requested: 47dc941dca801b89f1529e09f0d338755d8fbb0ce75f526e1dcccc4fbf8fc66c) - deleting download.
> + curl -f --connect-timeout 20 --retry 5 --location --insecure http://sources.lede-project.org/usb.ids
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100  561k  100  561k    0     0   164k      0  0:00:03  0:00:03 --:--:--  164k
> Hash of the downloaded file does not match (file: 6a39eb0ec15fa6235ede06bbfc39d54fa01bcea3ef44cb69cab9abff11997fde, requested: 47dc941dca801b89f1529e09f0d338755d8fbb0ce75f526e1dcccc4fbf8fc66c) - deleting download.
> + curl -f --connect-timeout 20 --retry 5 --location --insecure http://mirror2.openwrt.org/sources/usb.ids
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
> curl: (22) The requested URL returned error: 404 Not Found
> Download failed.
> + curl -f --connect-timeout 20 --retry 5 --location --insecure http://downloads.openwrt.org/sources/usb.ids
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100  466k  100  466k    0     0   265k      0  0:00:01  0:00:01 --:--:--  264k
> Hash of the downloaded file does not match (file: 312d411c21404a16516cb4d33dcda4cb1c7ed43c003a8736db274de655e418d7, requested: 47dc941dca801b89f1529e09f0d338755d8fbb0ce75f526e1dcccc4fbf8fc66c) - deleting download.
> No more mirrors to try - giving up.
> make[3]: *** [/home/lede/lede/dl/usb.ids] Error 2
> make[3]: Leaving directory `/home/lede/lede/feeds/packages/net/usbip'
> make[2]: *** [package/feeds/packages/usbip/compile] Error 2

hxy8241, 2017-11-30:
这是网络问题吧？全局翻墙试试看。

badsirius, 2017-11-30:
在Azure上编译的，不存在这个问题。
我根据提示，自己wget下载了usb.ids ( http://www.linux-usb.org/usb.ids ）
然后自己计算下载到的文件的md5 (md5sum usb.ids)，
把计算的结果替换Makefile里的数值后，编译成功。

hxy8241, 2017-11-30:
挺奇怪的，我编译时没这个问题。

closed 2017-12-02 by badsirius