coolsnowwolf/lede #219: 斐讯K3 IPv6 DHCP Relay问题

PaTTeeL, 2017-12-08:
wan口使用PPPoE拨号获取公网IP，wan6支持6to4（IPv6 over IPv4）获取（IPv6: 2002:65e3:b12e::1/16、IPv6-PD: 2002:65e3:b12e::/48），这时lan的信息是（IPv4: 192.168.2.1/24、IPv6: 2002:65e3:b12e::1/60），试过所有的组合（IPv6 ULA 前缀有和没、IPv6 分配长度60和64、DHCP服务器IPv6设置中的路由通告服务、DHCPv6 服务、NDP 代理改成中继模式、混合模式和服务器模式）局域网客户端都无法获取IPv6地址，请教如何设置和解决？用的1205版本的LEDE

LGA1150, 2017-12-20:
如果你安装了 mwan3 请参考 openwrt/packages#2429 修改配置文件

PaTTeeL, 2017-12-21:
多谢，已解决问题，是odhcpd包缺失的问题，之前没有装，无论怎么调整都无效，现在装好了odhcpd，然后把路由通告服务和dhcpv6都设为服务器，ndp设为中继，所有pc的ipv6问题都搞定了。BTW，未启用mwan3设置。

LGA1150, 2017-12-22:
这种情况下没必要打开 ndp 中继

PaTTeeL, 2017-12-22:
我现在是用的he.net的6in4，固定的ipv6地址，然后获得了一个/64的网段，内网的ipv6获取还需要开ndp中继吗？另外在哪里输入he.net或者google的ipv6 dns服务器呢？interface里找不到配置的地方。我看了梅林的ipv6配置页面，是可以填入ipv6的dns服务器的，openwrt/lede里面找不到对应的地方，请教填哪里？

pchaos, 2017-12-23:
我也是he.net的6in4，本地pc获取不到v6地址

pchaos, 2017-12-23:
有支持内网的v6的config.seed文件能共享一下吗？

LGA1150, 2017-12-24:
不建议直接修改 DNS，建议使用 DNS 分线路解析代理工具如 overture、Pcap DNSProxy、ChinaDNS

cnritng, 2018-03-15:
我也是he的6in4,但是路由器都获取不到ipv6地址,请教各位怎么弄?

LGA1150, 2018-03-22:
@cnritng 
确保 6in4 kmod-sit 已安装

closed 2018-10-06 by coolsnowwolf