coolsnowwolf/lede #233: [bug]Transmission端口无法

PaTTeeL, 2017-12-17:
最新版的Transmission端口51413一直显示关闭，无法打开，无法连接PT服务器，尝试过openssl和mbedtls两个2.92的版本，安装了所有的软件包（luci、daemon、cli和remote），防火墙已在流量规则中添加开放端口“Transmission | 任何 tcp, udp | 来自 所有主机 位于 wan | 到 所有路由IP 在 port 51413 位于 本设备，以及在自定义规则添加了iptables -I INPUT -p udp --dport 51413 -j ACCEPT和iptables -I INPUT -p tcp --dport 51413 -j ACCEPT，依然在transmission的web ui中显示Port is Closed，已下载的种子和文件可以读取、校验和加载，但上传下载速度都是0，特来请教如何除虫。
网络情况，上海电信PPPoE，手动通过6in4在He.net上获取的公网IPv6地址，pc上的ut2.21通过UPnP正常连接PT服务器且能跑满带宽。

PaTTeeL, 2017-12-21:
补充：问题已经解决。
解决办法：使用/lede/releases/17.01.4/packages/arm_cortex-a9/packages下transmission的ipk包（版本号为2.92-git4）安装后解决pt连接失败问题。反复尝试确认snapshot下的ipk包（版本号为2.92-git5）安装后无法连接PT服务器同时无法接受外部peer的连接（即使关闭防火墙也显示port closed）

leopoo, 2017-12-25:
同样问题啊. 但是装上2.92-git4 无法解决啊?

PaTTeeL, 2017-12-25:
@leopoo 交代一下具体情况，装了哪里下载的tr，系统怎么报错？有没有log贴一下，另外还有做过哪些尝试？

leopoo, 2017-12-26:
软件源
`
src/gz openwrt_core http://downloads.lede-project.org/releases/17.01.4/targets/bcm53xx/generic/packages

src/gz openwrt_base http://downloads.lede-project.org/releases/17.01.4/packages/arm_cortex-a9/base

src/gz openwrt_luci http://downloads.lede-project.org/releases/17.01.4/packages/arm_cortex-a9/luci

src/gz openwrt_packages http://downloads.lede-project.org/releases/17.01.4/packages/arm_cortex-a9/packages

src/gz openwrt_routing http://downloads.lede-project.org/releases/17.01.4/packages/arm_cortex-a9/routing

src/gz openwrt_telephony http://downloads.lede-project.org/releases/17.01.4/packages/arm_cortex-a9/telephony
`
版本
软件名 | 版本
-- | --
主机名 | LEDE-K3
主机型号 | PHICOMM K3
固件版本 | OpenWrt 17.01.4 R7.5.1 By Lean / LuCI Master (git-17.328.04231-802d5b6)
 luci-app-transmission | git-17.336.23170-d2dc32a-1
luci-i18n-transmission-zh-cn | git-17.336.23170-d2dc32a-1
 transmission-daemon-openssl | 2.92+git-4
 transmission-remote-openssl | 2.92+git-4
transmission-web | 2.92+git-4

日志

系统日志里有什么输出啊. 我把TR message_level 设置成3 也没有 是打印在其他地方?

`
Tue Dec 26 19:15:41 2017 authpriv.info dropbear[29299]: Child connection from 192.168.1.173:59656
Tue Dec 26 19:15:41 2017 authpriv.notice dropbear[29299]: Password auth succeeded for 'root' from 192.168.1.173:59656
Tue Dec 26 19:15:53 2017 authpriv.info dropbear[29605]: Child connection from 192.168.1.173:59657
Tue Dec 26 19:15:54 2017 authpriv.notice dropbear[29605]: Password auth succeeded for 'root' from 192.168.1.173:59657
Tue Dec 26 19:16:20 2017 user.notice transmission: Starting with 257702000 virt mem
Tue Dec 26 19:21:48 2017 user.notice transmission: Starting with 257702000 virt mem
Tue Dec 26 19:21:48 2017 user.notice transmission: Starting with 257702000 virt mem
Tue Dec 26 19:26:28 2017 user.notice transmission: Starting with 257702000 virt mem
Tue Dec 26 19:26:46 2017 kern.err /usr/bin/ssr-redir[20604]: server recv: Connection reset by peer
Tue Dec 26 19:26:48 2017 kern.err /usr/bin/ssr-redir[20604]: server recv: Connection reset by peer
Tue Dec 26 19:29:27 2017 kern.err /usr/bin/ssr-redir[20604]: server recv: Connection reset by peer
Tue Dec 26 19:39:40 2017 kern.err /usr/bin/ssr-redir[20604]: server recv: Connection reset by peer

`

leopoo, 2017-12-26:
貌似找到原因了. 电法分配给我的是内网IP...

closed 2018-10-06 by coolsnowwolf