coolsnowwolf/lede #245: 请问编译出错这个是什么问题？

ivyfm2, 2018-01-03:
请问编译出错这个是什么问题？
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-autoreboot:
 * 	libuci-lua (>= 2018-01-01) * 
 * opkg_install_cmd: Cannot install package luci-app-autoreboot.
make[2]: *** [package/install] Error 255
make[2]: Leaving directory `/home/fm2/lede'
make[1]: *** [/home/fm2/lede/staging_dir/target-mips_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory `/home/fm2/lede'
make: *** [world] 错误 2

tefiszx, 2018-01-05:
make menuconfig 的时候可能依赖不完整。试试看把luci-app-autoreboot功能除去后再编译。

ivyfm2, 2018-01-06:
谢谢，把autoreboot功能去掉还是会出现这个•libuci-lua (>= 2018-01-01) *错误

我用了git pull这个命令，然后重新feeds update后居然莫名其妙的好了！

closed 2018-10-06 by coolsnowwolf