coolsnowwolf/lede #247: 编译X86出错了，请帮忙看看

zxjack, 2018-01-06:
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-lib-fs:
 * 	luci-proto-ipv6 * 
 * opkg_install_cmd: Cannot install package luci-lib-fs.
 * check_data_file_clashes: Package bridge wants to install file /home/zxjack/lede/build_dir/target-x86_64_musl/root-x86/usr/sbin/brctl
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package bridge.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for strongswan:
 * 	kmod-ipsec6 * 
 * opkg_install_cmd: Cannot install package strongswan.
make[2]: *** [package/install] Error 255
make[2]: Leaving directory `/home/zxjack/lede'
make[1]: *** [/home/zxjack/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory `/home/zxjack/lede'

zxjack, 2018-01-06:
Configuring luci-i18n-firewall-zh-cn.
Collected errors:
 * check_data_file_clashes: Package bridge wants to install file /home/zxjack/lede/build_dir/target-x86_64_musl/root-x86/usr/sbin/brctl
	But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package bridge.
make[2]: *** [package/install] Error 255
make[2]: Leaving directory `/home/zxjack/lede'
make[1]: *** [/home/zxjack/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory `/home/zxjack/lede'
make: *** [world] Error 2

Moekr, 2018-01-06:
busybox的brctl和bridge-utils的brctl二选一

zxjack, 2018-01-06:
多谢了，我试试看

closed 2018-10-06 by coolsnowwolf

qiuxinpeng, 2020-02-15:
> busybox的brctl和bridge-utils的brctl二选一

在哪里选择呢,我没找到二选一的选项

ElmerZhang, 2020-02-20:
> > busybox的brctl和bridge-utils的brctl二选一
> 
> 在哪里设置选择呢,我找不到在哪里修改,让编译通过

make menuconfig -> Base system -> bridge 和 busybox 二选一

liang747, 2021-02-23:
> > > busybox的brctl和bridge-utils的brctl二选一
> > 
> > 在哪里设置选择呢,我找不到在哪里修改,让编译通过
> 
> make menuconfig -> Base system -> bridge 和 busybox 二选一

我在这里做了二选一但是编译时还是报错我选了busybox因为我需要挂载nfs，而brigde一直默认是关闭的。