coolsnowwolf/lede #257: R7.5.2 编译提示 ODHCPD 错误

anglabace, 2018-01-10:
Configuring luci-i18n-upnp-zh-cn.
Configuring 6rd.
Configuring luci-i18n-firewall-zh-cn.
Configuring 6to4.
Configuring ppp-mod-pppoe.
Collected errors:
 * check_data_file_clashes: Package odhcpd-ipv6only wants to install file /home/anglabace/ipqlede/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/init.d/odhcpd
	But that file is already provided by package  * odhcpd
 * check_data_file_clashes: Package odhcpd-ipv6only wants to install file /home/anglabace/ipqlede/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/uci-defaults/odhcpd.defaults
	But that file is already provided by package  * odhcpd
 * check_data_file_clashes: Package odhcpd-ipv6only wants to install file /home/anglabace/ipqlede/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/sbin/odhcpd
	But that file is already provided by package  * odhcpd
 * check_data_file_clashes: Package odhcpd-ipv6only wants to install file /home/anglabace/ipqlede/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/usr/sbin/odhcpd-update
	But that file is already provided by package  * odhcpd
 * opkg_install_cmd: Cannot install package odhcpd-ipv6only.
make[2]: *** [package/install] Error 255
make[2]: Leaving directory `/home/anglabace/ipqlede/lede'
make[1]: *** [/home/anglabace/ipqlede/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory `/home/anglabace/ipqlede/lede'
make: *** [world] 错误 2
anglabace@anglabace-virtual-machine:~/ipqlede/lede$

ctw8989, 2018-01-10:
在Network里取消odhcpd

anglabace, 2018-01-10:
@ctw8989  然而我用ipv6 所以不能取消。

Moekr, 2018-01-10:
有odhcpd为什么还要选odhcpd-ipv6only？

anglabace, 2018-01-10:
@Moekr  貌似是自动关联上， odhcpd 这里选上ipv6 目的我认为这里只为v6 服务而不需要为v4服务。v4交给dnsmasq 处理。

Moekr, 2018-01-10:
@anglabace 我使用NAT6方案，只安装了odhcpd没有odhcpd-ipv6only，v6使用一切正常，我觉得这里不需要选上odhcpd-ipv6only，即便是自动关联的

anglabace, 2018-01-10:
@Moekr 然而我貌似没找到关闭 odhcpd-ipv6only 选项

coolsnowwolf, 2018-01-15:
去掉了

closed 2018-10-06 by coolsnowwolf