coolsnowwolf/lede #262: 编译出错，请教，感谢了。

jxph1123, 2018-01-11:
编译了两次都出现下面错误，求教了感谢！checking pcre/pcre.h presence... no
checking for pcre/pcre.h... no
checking for asciidoc... no
configure: error: Cannot find `asciidoc` in PATH.
Makefile:224: recipe for target '/home/etek/lede/build_dir/target-mipsel_24kc_musl/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/etek/lede/build_dir/target-mipsel_24kc_musl/shadowsocksR-libev-openssl/shadowsocksR-libev-2.5.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/etek/lede/package/lean/shadowsocksR-libev-full'
package/Makefile:105: recipe for target 'package/lean/shadowsocksR-libev-full/compile' failed
make[2]: *** [package/lean/shadowsocksR-libev-full/compile] Error 2
make[2]: Leaving directory '/home/etek/lede'
package/Makefile:101: recipe for target '/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/etek/lede'
/home/etek/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

jxph1123, 2018-01-11:
全局翻墙，只改了型号，小米3g，其他都是默认配置

daofeng2015, 2018-01-13:
编译ssr依赖增多了 你这是差环境

AndyX-Net, 2018-01-13:
configure: error: Cannot find **asciidoc** in PATH
启用酸酸乳需要额外组件

执行以下语句：
sudo apt-get update
sudo apt-get install asciidoc

大约需要1.5GB的空间

closed 2018-10-06 by coolsnowwolf