coolsnowwolf/lede #274: R7.5.4，在make menuconfig里找不到自定义包

fsyxhua, 2018-01-17:
在package/lean里的包都在make menuconfig找不到，请教什么原因

allenfoxtc, 2018-01-17:
在LuCI -> 3. Applications里
如果没有 先执行下
```
./scripts/feeds update -a 
./scripts/feeds install -a
```

fsyxhua, 2018-01-18:
肯定是执行了，问题跟#276一样的，应该是官方feed源出问题了，我看openwrt官方也有类似问题出现

fsyxhua, 2018-01-18:
用 feeds update -a更新包，remote: Counting objects: **# 3851**, done.
[fsyx@localhost lean-lede]$ ./scripts/feeds update -a
Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git;lede-17.01' ...
Cloning into './feeds/packages'...
remote: Counting objects: 3851, done.
remote: Compressing objects: 100% (3263/3263), done.
remote: Total 3851 (delta 153), reused 3021 (delta 78)
Receiving objects: 100% (3851/3851), 2.37 MiB | 156.00 KiB/s, done.
Resolving deltas: 100% (153/153), done.
Create index file './feeds/packages.index' 
Collecting package info: done
Collecting target info: done
==========================================
用git命令直接clone就有，remote: Counting objects: **55048**, done.
[fsyx@localhost src]$ git clone -b lede-17.01 https://git.openwrt.org/feed/packages.git
Cloning into 'packages'...
remote: Counting objects: 55048, done.
remote: Compressing objects: 100% (25283/25283), done.
remote: Total 55048 (delta 29617), reused 51392 (delta 26644)
Receiving objects: 100% (55048/55048), 13.38 MiB | 315.00 KiB/s, done.
Resolving deltas: 100% (29617/29617), done.

closed 2018-10-06 by coolsnowwolf