coolsnowwolf/lede #276: 编译出错，有点长.....

ziqqqq, 2018-01-17:
WARNING: Makefile 'package/feeds/packages/coova-chilli/Makefile' has a dependency on 'libmatrixssl', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsqlite2', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'flite', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libyuv', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'hiredis', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libilbc', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libmemcached', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsilk', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libg7221', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'freeradius-client', which does not exist
WARNING: Makefile 'package/feeds/luci/freifunk-p2pblock/Makefile' has a dependency on 'l7-protocols', which does not exist
WARNING: Makefile 'package/feeds/packages/libiio/Makefile' has a dependency on 'libavahi-common', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-usb-musb-tusb6010', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'smap', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'mac-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'smap-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'netdiscover-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-authenticate', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-disa', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-setcallerid', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-system', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-chan-gtalk', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-codec-a-mu', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-codec-alaw', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-func-cut', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-res-clioriginate', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-func-channel', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-chan-local', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-record', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-senddtmf', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-res-crypto', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx-voicemail/Makefile' has a dependency on 'asterisk18', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-radvd/Makefile' has a dependency on 'radvd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-ushare/Makefile' has a dependency on 'ushare', which does not exist
make[2]: Entering directory '/home/zh/lede/scripts/config'
make[2]: Leaving directory '/home/zh/lede/scripts/config'
make[1]: Entering directory '/home/zh/lede'
make[2]: Entering directory '/home/zh/lede'
+ mkdir -p /home/zh/lede/staging_dir/target-mips_24kc_musl
+ cd /home/zh/lede/staging_dir/target-mips_24kc_musl
+ mkdir -p bin lib include stamp
mkdir -p /home/zh/lede/build_dir/target-mips_24kc_musl/stamp
touch /home/zh/lede/staging_dir/target-mips_24kc_musl/.prepared
make[3]: Entering directory '/home/zh/lede/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/zh/lede/toolchain/gdb'
make[3]: Entering directory '/home/zh/lede/toolchain/binutils'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/zh/lede/toolchain/binutils'
make[3]: Entering directory '/home/zh/lede/toolchain/gcc/initial'
export SHELL="bash"; make CFLAGS="-O2 -I/home/zh/lede/staging_dir/host/include -I/home/zh/lede/staging_dir/host/usr/include " CFLAGS_FOR_TARGET="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float" CXXFLAGS_FOR_TARGET="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float" GOCFLAGS_FOR_TARGET="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float"  -C /home/zh/lede/build_dir/toolchain-mips_24kc_gcc-5.5.0_musl/gcc-5.5.0-initial all-build-libiberty all-gcc all-target-libgcc
make[4]: Entering directory '/home/zh/lede/build_dir/toolchain-mips_24kc_gcc-5.5.0_musl/gcc-5.5.0-initial'
Configuring in build-x86_64-pc-linux-gnu/libiberty
make[5]: Entering directory '/home/zh/lede/build_dir/toolchain-mips_24kc_gcc-5.5.0_musl/gcc-5.5.0-initial/build-x86_64-pc-linux-gnu/libiberty'
make[5]: *** No rule to make target 'all'.  Stop.
make[5]: Leaving directory '/home/zh/lede/build_dir/toolchain-mips_24kc_gcc-5.5.0_musl/gcc-5.5.0-initial/build-x86_64-pc-linux-gnu/libiberty'
Makefile:2572: recipe for target 'all-build-libiberty' failed
make[4]: *** [all-build-libiberty] Error 2
make[4]: Leaving directory '/home/zh/lede/build_dir/toolchain-mips_24kc_gcc-5.5.0_musl/gcc-5.5.0-initial'
Makefile:36: recipe for target '/home/zh/lede/build_dir/toolchain-mips_24kc_gcc-5.5.0_musl/gcc-5.5.0-initial/.built' failed
make[3]: *** [/home/zh/lede/build_dir/toolchain-mips_24kc_gcc-5.5.0_musl/gcc-5.5.0-initial/.built] Error 2
make[3]: Leaving directory '/home/zh/lede/toolchain/gcc/initial'
toolchain/Makefile:96: recipe for target 'toolchain/gcc/initial/compile' failed
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/zh/lede'
toolchain/Makefile:94: recipe for target '/home/zh/lede/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/zh/lede/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/zh/lede'
/home/zh/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

fsyxhua, 2018-01-18:
应该是官方的feed源出了问题，我的也这样，感觉丢失了很多包

edidada, 2018-01-20:
同样遇到类似的问题，cenetos 7，按照官方的教程来的

clistant, 2018-01-21:
同样的问题，前几个月编译还正常。

ziqqqq, 2018-01-21:
我也是，前两天还好好的，这两天就这样了，清理以后还是这个问题。

closed 2018-10-06 by coolsnowwolf