coolsnowwolf/lede #278: 编译中遇到问题，哪位帮忙解决一下，感谢

arthaswu83, 2018-01-18:
~/lede$ ./scripts/feeds install -a
Collecting package info: done
Collecting target info: done
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'ntfs-3g', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
运行到这部提示错误，请问我该如何操作

plutoghost, 2018-01-18:
feeds update 
feeds install -a

arthaswu83, 2018-01-18:
非常感谢

| |

邮箱：arthaswu83@163.com
|

签名由 网易邮箱大师 定制

在2018年01月18日 20:05，plutoghost 写道：

feeds update
feeds install -a

edidada, 2018-01-20:
WARNING: Makefile 'package/feeds/packages/coova-chilli/Makefile' has a dependency on 'libmatrixssl', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsqlite2', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'flite', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libyuv', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'hiredis', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libilbc', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libmemcached', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsilk', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libg7221', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'freeradius-client', which does not exist
WARNING: Makefile 'package/feeds/luci/freifunk-p2pblock/Makefile' has a dependency on 'iptables-mod-ipp2p', which does not exist
WARNING: Makefile 'package/feeds/luci/freifunk-p2pblock/Makefile' has a dependency on 'l7-protocols', which does not exist
WARNING: Makefile 'package/feeds/packages/libiio/Makefile' has a dependency on 'libavahi-common', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'smap', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'mac-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'smap-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'netdiscover-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-authenticate', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-disa', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-setcallerid', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-system', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-chan-gtalk', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-codec-a-mu', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-codec-alaw', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-func-cut', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-res-clioriginate', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-func-channel', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-chan-local', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-record', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-senddtmf', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-res-crypto', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx-voicemail/Makefile' has a dependency on 'asterisk18', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-radvd/Makefile' has a dependency on 'radvd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-ushare/Makefile' has a dependency on 'ushare', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.

closed 2018-10-06 by coolsnowwolf