coolsnowwolf/lede #280: X86-64编译，编译时出错

anglabace, 2018-01-19:
export SHELL="bash"; make CFLAGS="-O2 -I/home/anglabace/x86lede/lede/staging_dir/host/include -I/home/anglabace/x86lede/lede/staging_dir/host/usr/include " CFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result" CXXFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result" GOCFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result"  -C /home/anglabace/x86lede/lede/build_dir/toolchain-x86_64_gcc-5.4.0_musl/gcc-5.4.0-initial prefix="/home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/initial" install-gcc install-target-libgcc
make[4]: Entering directory `/home/anglabace/x86lede/lede/build_dir/toolchain-x86_64_gcc-5.4.0_musl/gcc-5.4.0-initial'
/bin/bash /home/anglabace/x86lede/lede/build_dir/toolchain-x86_64_gcc-5.4.0_musl/gcc-5.4.0/mkinstalldirs /home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/initial /home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/initial
make[5]: Entering directory `/home/anglabace/x86lede/lede/build_dir/toolchain-x86_64_gcc-5.4.0_musl/gcc-5.4.0-initial/fixincludes'
rm -rf /home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/initial/libexec/gcc/x86_64-openwrt-linux-musl/5.4.0/install-tools
/bin/bash /home/anglabace/x86lede/lede/build_dir/toolchain-x86_64_gcc-5.4.0_musl/gcc-5.4.0/fixincludes/../mkinstalldirs /home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/initial/libexec/gcc/x86_64-openwrt-linux-musl/5.4.0/install-tools
mkdir -p -- /home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/initial/libexec/gcc/x86_64-openwrt-linux-musl/5.4.0/install-tools
mkdir: cannot create directory '/home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/initial/libexec/gcc/x86_64-openwrt-linux-musl/5.4.0/install-tools': No space left on device
make[5]: *** [install] Error 1
make[5]: Leaving directory `/home/anglabace/x86lede/lede/build_dir/toolchain-x86_64_gcc-5.4.0_musl/gcc-5.4.0-initial/fixincludes'
make[4]: *** [install-fixincludes] Error 2
make[4]: Leaving directory `/home/anglabace/x86lede/lede/build_dir/toolchain-x86_64_gcc-5.4.0_musl/gcc-5.4.0-initial'
make[3]: *** [/home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/stamp/.gcc_initial_installed] Error 2
make[3]: Leaving directory `/home/anglabace/x86lede/lede/toolchain/gcc/initial'
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory `/home/anglabace/x86lede/lede'
make[1]: *** [/home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory `/home/anglabace/x86lede/lede'
make: *** [world] 错误 2
anglabace@anglabace-virtual-machine:~/x86lede/lede$

Moekr, 2018-01-19:
```
mkdir: cannot create directory '/home/anglabace/x86lede/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl/initial/libexec/gcc/x86_64-openwrt-linux-musl/5.4.0/install-tools': No space left on device
```
硬盘满了。。。编译LEDE最好留出50G以上的可用空间

anglabace, 2018-01-20:
@Moekr 实际还有30GB空间是空闲的。。

closed 2018-10-06 by coolsnowwolf