coolsnowwolf/lede #287: R755 R7800编译提示错误

anglabace, 2018-01-24:
cp /home/anglabace/bcmlede/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/tmp/openwrt-ipq806x-netgear_r7800-squashfs-factory.img /home/anglabace/bcmlede/lede/bin/targets/ipq806x/generic/openwrt-ipq806x-netgear_r7800-squashfs-factory.img
/home/anglabace/bcmlede/lede/staging_dir/host/bin/mkfs.ubifs   --space-fixup   --compr=zlib  --squash-uids -o /home/anglabace/bcmlede/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/root.ubifs -d /home/anglabace/bcmlede/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x
Error: min. I/O unit was not specified (use -h for help)
make[5]: *** [/home/anglabace/bcmlede/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/root.ubifs] Error 255
make[5]: Leaving directory `/home/anglabace/bcmlede/lede/target/linux/ipq806x/image'
make[4]: *** [install] Error 2
make[4]: Leaving directory `/home/anglabace/bcmlede/lede/target/linux/ipq806x'
make[3]: *** [install] Error 2
make[3]: Leaving directory `/home/anglabace/bcmlede/lede/target/linux'
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory `/home/anglabace/bcmlede/lede'
make[1]: *** [/home/anglabace/bcmlede/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory `/home/anglabace/bcmlede/lede'
Build failed - please re-run with -j1 to see the real error message
make: *** [world] 错误 1
anglabace@anglabace-virtual-machine:~/bcmlede/lede$

sirfromprc, 2018-01-25:
EA8500 也一样提示，出现在最后一步无法完成。但是型号换成ac58u，又可以编译成功

sirfromprc, 2018-01-25:
找到原因了，你make menuconfig 选targe images 关闭ubifs选项即可。

anglabace, 2018-01-25:
@sirfromprc  谢啦

closed 2018-10-06 by coolsnowwolf