coolsnowwolf/lede #291: 路由器选择怎么没有tplink941v2了？感觉少了好多老版的路由器，是不再支持了吗？

sdfgh2046, 2018-01-29:
假如是这样的话，怎么编译老版本？
新手，不太懂。谢谢。
编译出现以下警告，不知原因
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsqlite2', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'flite', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libyuv', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'hiredis', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libilbc', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libmemcached', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsilk', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libg7221', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'freeradius-client', which does not exist

daofeng2015, 2018-01-29:
不报错就不要管，这是有的包依赖没有才报错的

AndyX-Net, 2018-02-02:
1、关于设备丢失问题参见https://github.com/coolsnowwolf/lede/issues/294 其实设备没丢，只是选项变了
2、遇到一大堆WARNING解决方法：在feeds.config.default中屏蔽telephony，还有问题的话删除所有原来的代码，然后重新git clone整个LEAN版本。

closed 2018-10-06 by coolsnowwolf