coolsnowwolf/lede #293: make 报错

wifislax, 2018-01-29:
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
Makefile:142: recipe for target 'cmGeneratorTarget.o' failed
make[4]: *** [cmGeneratorTarget.o] Error 4
make[4]: Leaving directory '/root/lede/build_dir/host/cmake-3.9.3/Bootstrap.cmk'
---------------------------------------------
Error when bootstrapping CMake:
Problem while running make
---------------------------------------------
Log of errors: /root/lede/build_dir/host/cmake-3.9.3/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
Makefile:35: recipe for target '/root/lede/build_dir/host/cmake-3.9.3/.configured' failed
make[3]: *** [/root/lede/build_dir/host/cmake-3.9.3/.configured] Error 9
make[3]: Leaving directory '/root/lede/tools/cmake'
tools/Makefile:148: recipe for target 'tools/cmake/compile' failed
make[2]: *** [tools/cmake/compile] Error 2
make[2]: Leaving directory '/root/lede'
tools/Makefile:146: recipe for target '/root/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.tools_compile_yynyyyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynyynynnyyynny' failed
make[1]: *** [/root/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.tools_compile_yynyyyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynyynynnyyynny] Error 2
make[1]: Leaving directory '/root/lede'
/root/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

hcwhan, 2018-09-05:
请问有解决吗 我也是这个报错

hcym, 2018-09-06:
环境问题吧，我是用的实体机1804，编译lean的和op主分支的。最近一直比较顺手。反正有用没用的包多安装一些

hcwhan, 2018-09-06:
是内存问题 
报错的时候 我是用一个 512M 小内存 VPS 编的
今天在本机 配了一个8G虚拟机 可以编译

closed 2018-10-06 by coolsnowwolf