coolsnowwolf/lede #294: 额，很久没有同步代码了， 今天刚同步发现删了不少设备的支持啊。

AndyX-Net, 2018-01-29:
WR840N WR882N WR941N 等等都没了~
 delete mode 100644 target/linux/ar71xx/files/arch/mips/ath79/mach-maselink-ap2600i.c
 delete mode 100644 target/linux/ar71xx/files/arch/mips/ath79/mach-maselink-ap2600ifm.c
 delete mode 100644 target/linux/ar71xx/files/arch/mips/ath79/mach-mw316r-v1.c
 delete mode 100644 target/linux/ar71xx/files/arch/mips/ath79/mach-sgr-w500-ebi-fit-v3.c
 delete mode 100755 target/linux/ar71xx/files/arch/mips/ath79/mach-sgr-w500-n85b-v2.c
 delete mode 100644 target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wdr6500-v6.c
 delete mode 100644 target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wr802n-v2.c
 delete mode 100644 target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wr882n-v1.c
 delete mode 100755 target/linux/ar71xx/files/arch/mips/ath79/mach-tl-wr941n-v7.c

AndyX-Net, 2018-01-30:
抱歉问题已经解决，查询了target/linux/ar71xx/image/目录发现使用了tiny分项。。。

选择make menuconfig -> Subtarget ->Deviece with small flash选项之后在Target Profile中即可见。

但是之前自定义的882Nv1和941Nv7都没有了比较可惜，希望能再次添加上。

closed 2018-01-30 by AndyX-Net

amoboy, 2018-02-23:
请问编译出的来固件没有squashfs......也没有factory.bin.....求解

AndyX-Net, 2018-02-23:
@amoboy 查询你输出的错误信息，预计可能是包过大(错误信息中包含 **_TOO_** BIG 字样)。也就是说这经常发生于4M 大小的FLASH固件上，放不了太多东西。