coolsnowwolf/lede #296: amule仍然不能编译。git的最新版。Ubuntu  14 LTS x64系统下编译。

shenzt68, 2018-01-31:
unittests/tests/Makefile.am:51: warning: source file '$(top_srcdir)/src/libs/common/Format.cpp' is in a subdirectory,
unittests/tests/Makefile.am:51: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:51: warning: source file '$(top_srcdir)/src/libs/common/strerror_r.c' is in a subdirectory,
unittests/tests/Makefile.am:51: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:19: warning: source file '$(top_srcdir)/src/kademlia/utils/UInt128.cpp' is in a subdirectory,
unittests/tests/Makefile.am:19: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:19: warning: source file '$(top_srcdir)/src/libs/common/Format.cpp' is in a subdirectory,
unittests/tests/Makefile.am:19: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:19: warning: source file '$(top_srcdir)/src/libs/common/strerror_r.c' is in a subdirectory,
unittests/tests/Makefile.am:19: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/SafeFile.cpp' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/CFile.cpp' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/MemFile.cpp' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/kademlia/utils/UInt128.cpp' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/libs/common/StringFunctions.cpp' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/Tag.cpp' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/libs/common/Path.cpp' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/libs/common/Format.cpp' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:40: warning: source file '$(top_srcdir)/src/libs/common/strerror_r.c' is in a subdirectory,
unittests/tests/Makefile.am:40: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:25: warning: source file '$(top_srcdir)/src/libs/common/Format.cpp' is in a subdirectory,
unittests/tests/Makefile.am:25: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:25: warning: source file '$(top_srcdir)/src/libs/common/strerror_r.c' is in a subdirectory,
unittests/tests/Makefile.am:25: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:34: warning: source file '$(top_srcdir)/src/NetworkFunctions.cpp' is in a subdirectory,
unittests/tests/Makefile.am:34: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:34: warning: source file '$(top_srcdir)/src/LibSocket.cpp' is in a subdirectory,
unittests/tests/Makefile.am:34: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:34: warning: source file '$(top_srcdir)/src/libs/common/Format.cpp' is in a subdirectory,
unittests/tests/Makefile.am:34: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:34: warning: source file '$(top_srcdir)/src/libs/common/strerror_r.c' is in a subdirectory,
unittests/tests/Makefile.am:34: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:43: warning: source file '$(top_srcdir)/src/libs/common/Path.cpp' is in a subdirectory,
unittests/tests/Makefile.am:43: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:43: warning: source file '$(top_srcdir)/src/libs/common/StringFunctions.cpp' is in a subdirectory,
unittests/tests/Makefile.am:43: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:28: warning: source file '$(top_srcdir)/src/libs/common/StringFunctions.cpp' is in a subdirectory,
unittests/tests/Makefile.am:28: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:28: warning: source file '$(top_srcdir)/src/libs/common/Path.cpp' is in a subdirectory,
unittests/tests/Makefile.am:28: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:46: warning: source file '$(top_srcdir)/src/libs/common/Path.cpp' is in a subdirectory,
unittests/tests/Makefile.am:46: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:46: warning: source file '$(top_srcdir)/src/libs/common/StringFunctions.cpp' is in a subdirectory,
unittests/tests/Makefile.am:46: but option 'subdir-objects' is disabled
unittests/tests/Makefile.am:46: warning: source file '$(top_srcdir)/src/libs/common/TextFile.cpp' is in a subdirectory,
unittests/tests/Makefile.am:46: but option 'subdir-objects' is disabled
make[3]: *** [/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/aMule-2.3.2/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
make[3]: Leaving directory `/root/lede/package/lean/amule'
make[2]: *** [package/lean/amule/compile] Error 2
make[2]: Leaving directory `/root/lede'
make[1]: *** [/root/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2

tinnlux, 2018-02-09:
一样的问题。求解决

paldier, 2018-02-09:
目测你们自己的问题，即使降级到opcc也是可以正常编译的

closed 2018-10-06 by coolsnowwolf